JohnIrvine1433/ThemeMaster #7: [enhancement request] Add Update / Install All Options in theme installation and update menus.

AngelofWoe, 2021-07-19:
Request to add Update / Install All Themes options in theme installation and update menus.

JohnIrvine1433, 2021-07-19:
Implemented on the nightly branch.
@AngelofWoe if you want to give it a try and tell me if that's ok.

AngelofWoe, 2021-07-19:
Seems to work quite well.

An option to skip the confirmation prompts between each theme (the '<themename> has been successfully installed' > OK prompt) wouldn't hurt but this is already much more convenient when installing many themes, thank you. In the case of installations I can also see someone wanting the ability to select multiple specific themes for installing but not necessarily all of them, but I also know that'd require more work.

The ability to update all outdated themes at once is perfect.

JohnIrvine1433, 2021-07-20:
Ok, I've implemented the additional features on the nightly branch but it needs some testing. I'll probably push it on the master branch this WE. Still time to make few adjustments !

AngelofWoe, 2021-07-21:
Skipping the prompts between themes and the counter are working excellently.

Install / Update All only installs/updates the selected themes instead of all available themes, so it functions identically to selected at the moment and won't run unless items are selected.

Can we add these features to Uninstall/Reinstall themes as well? I did a lot of uninstalling while testing (although I suppose I could have just deleted the files again).

JohnIrvine1433, 2021-07-21:
Thanks for your feedback, I'll tqke account of it before releasing the version.

AngelofWoe, 2021-07-21:
Thank you so much for all your hard work.

JohnIrvine1433, 2021-07-24:
@AngelofWoe, having gone through it I decided that it might be better to leave a simple list for Uninstall and Reinstall menus. Usually you don't intend to remove or reinstall several themes at once and it's way faster to remove one with a simple list than with a checklist... as you said during tests you can always delete the files.
Otherwise everything should be working smoothly now.

JohnIrvine1433, 2021-07-25:
Version 4.0.0 now released !

closed 2021-07-25 by JohnIrvine1433