JohnIrvine1433/ThemeMaster #8: ES Startup Stuck in "Preloading UI" Loop

weirdeetz, 2021-09-15:
Hi John,
I'm using ArkOS on my RG351P (latest firmware).
I've install ThemeMaster and download all provided themes to test it one by one.
I start testing the first theme (ArkOS-Carbon) and everything is fine..
But then I choose the second theme (Art-Book) and ES restarted, something bad happen.
My ES Startup Stuck in "Preloading UI" Loop.
I've tried deleting the Art-Book folder, and it still wont fixed.
I've also tried to duplicate my existing theme (EpicNoir) and rename it as "Art-Book" but the result is still the same.
I can't seem to get past this never-ending loop on start up.
Please help :(

Best,
-Adit

JohnIrvine1433, 2021-09-16:
Have you tried removing all added themes (delete all added folders)?
Please let me know what happens. 
I'll also check with the ArkOS guy about the Preloading UI option as I cannot find the configuration file to change it on ArkOS.

weirdeetz, 2021-09-16:
I did. I even removed all themes and replace it with all original themes I get from freshly installed ArkOS (I already made a backup before) the startup still looping.
And the even weird thing is: I desperately deleted all files inside BOOT and EASYROMS drives in my 128 Sandisk, replace it with files from my old original Anbernic microSD card... and still... won't booting up, still looping.

So I assumed it was error my Sandisk card.
Then I just format my Sandisk microSD card again last night..
Put everything back from my backup because I'm to lazy spending any more hours to set everything.
And everything back to normal! \(^.^)/

Yes so... I think it was issue on my microSD, not your ThemeMaster.

[SOLVED]
Thank you very much for your attention, keep up doing your great job man

JohnIrvine1433, 2021-09-16:
Good to hear everything is back to normal. ThemeMaster should not directly cause ES to break but sometimes you may end end with ES crashing because of a theme... in that case removing it should have solved your problem ! 
Always good to have a back up somewhere :)

closed 2021-09-16 by JohnIrvine1433