JohnIrvine1433/ThemeMaster #13: Will the Powkiddy X55 be supported?

gannon345, 2023-05-23:
Powkiddy X55 runs JELOS stock, but isn't supported due to a lack of control mapping (though thememaster does launch via the JELOS instructions).

slaminger, 2023-05-26:
> Unfortunately I do not have this plateform, however I can implement the
> required changes if you can provide me with the required information (input
> file for X55 controls and event codes)

Does it support Xbox 360 controllers?

gannon345, 2023-05-27:
@JohnIrvine1433 do you know where I would find the files on the device and what they might be names? I’ll be happy to provide it, just not sure what to look for.

JohnIrvine1433, 2023-05-28:
@gannon345 it should be under "/dev/input/by-path/" ; regarding controls event codes, you can get them by running evtest.

gannon345, 2023-05-28:
I'm having a hard time getting the files. The files in ../by-path are:

platform-fdd40000.i2c-platform-rk805-pwrkey-event
platform-rk-headset-event
platform-zed_joystick-event
platform-zed_joystick-event-joystick
platform-zed_joystick-joystick

But they are all symlinks to event files in /dev/input:

event0
event1
event2
event3
js0

All of them are blank files, and they're not simlinks. It's very confusing. And I wasn't able to scp any of the files off, they all errored. 

I was able to run evtest though, and output comes from event0 (zed_joystick). For all the digital buttons (L1-L3, R1-R3, face buttons, dpad, start, select) 1 is pressed, 0 is released. Not sure how to read the joystick values.

Results below:

evtest for zed_joystick:

Available devices:
/dev/input/event0:	zed_joystick
/dev/input/event1:	rk805 pwrkey
/dev/input/event2:	zed_keyboard
/dev/input/event3:	rk-headset
Select the device event number [0-3]: 0
Input driver version is 1.0.1
Input device ID: bus 0x19 vendor 0x0 product 0x0 version 0x0
Input device name: "zed_joystick"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 304 (BTN_SOUTH)
    Event code 305 (BTN_EAST)
    Event code 307 (BTN_NORTH)
    Event code 308 (BTN_WEST)
    Event code 310 (BTN_TL)
    Event code 311 (BTN_TR)
    Event code 312 (BTN_TL2)
    Event code 313 (BTN_TR2)
    Event code 314 (BTN_SELECT)
    Event code 315 (BTN_START)
    Event code 316 (BTN_MODE)
    Event code 317 (BTN_THUMBL)
    Event code 318 (BTN_THUMBR)
    Event code 544 (BTN_DPAD_UP)
    Event code 545 (BTN_DPAD_DOWN)
    Event code 546 (BTN_DPAD_LEFT)
    Event code 547 (BTN_DPAD_RIGHT)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value      9
      Min        0
      Max       18
      Flat       9
    Event code 1 (ABS_Y)
      Value      9
      Min        0
      Max       18
      Flat       9
    Event code 3 (ABS_RX)
      Value      9
      Min        0
      Max       18
      Flat       9
    Event code 4 (ABS_RY)
[... 87 lines cut ...]
Event: time 1685309815.289010, -------------- SYN_REPORT ------------
Event: time 1685309815.389052, type 3 (EV_ABS), code 4 (ABS_RY), value 9
Event: time 1685309815.389052, -------------- SYN_REPORT ------------
Event: time 1685309815.722215, type 3 (EV_ABS), code 4 (ABS_RY), value 18
Event: time 1685309815.722215, -------------- SYN_REPORT ------------
Event: time 1685309815.855654, type 3 (EV_ABS), code 4 (ABS_RY), value 9
Event: time 1685309815.855654, -------------- SYN_REPORT ------------
Event: time 1685309816.222295, type 3 (EV_ABS), code 3 (ABS_RX), value 0
Event: time 1685309816.222295, -------------- SYN_REPORT ------------
Event: time 1685309816.292267, type 3 (EV_ABS), code 3 (ABS_RX), value 9
Event: time 1685309816.292267, -------------- SYN_REPORT ------------
Event: time 1685309816.662198, type 3 (EV_ABS), code 3 (ABS_RX), value 12
Event: time 1685309816.662198, -------------- SYN_REPORT ------------
Event: time 1685309816.698834, type 3 (EV_ABS), code 3 (ABS_RX), value 18
Event: time 1685309816.698834, -------------- SYN_REPORT ------------
Event: time 1685309816.802198, type 3 (EV_ABS), code 3 (ABS_RX), value 9
Event: time 1685309816.802198, -------------- SYN_REPORT ------------
Event: time 1685309818.418720, type 1 (EV_KEY), code 314 (BTN_SELECT), value 1
Event: time 1685309818.418720, -------------- SYN_REPORT ------------
Event: time 1685309818.545521, type 1 (EV_KEY), code 314 (BTN_SELECT), value 0
Event: time 1685309818.545521, -------------- SYN_REPORT ------------
Event: time 1685309819.157072, type 1 (EV_KEY), code 315 (BTN_START), value 1
Event: time 1685309819.157072, -------------- SYN_REPORT ------------
Event: time 1685309819.232143, type 1 (EV_KEY), code 315 (BTN_START), value 0
Event: time 1685309819.232143, -------------- SYN_REPORT ------------

JohnIrvine1433, 2023-06-03:
@gannon345 an updated version has been push to the "x55" branch. If you can give it a try and let me know if it works properly...
https://github.com/JohnIrvine1433/ThemeMaster/archive/refs/heads/x55.zip

SuperBadger77, 2023-06-03:
I was just about to raise an issue to see if you would add support for the X55. I'll also give this a go and report back.

SuperBadger77, 2023-06-03:
So I copied it across and ran it. Previously I got into the same issue as @gannon345 where it loaded but said there was a lack of controls mapped and had to connect a USB keyboard to it to get it to run. Now it gets further but I get the following error message (pic attached). Again no controls worked so I connected my usb keyboard and selected continue and it bombed out back to the main menu. Attached the error message as well.

[es_launch_stdout.log](https://github.com/JohnIrvine1433/ThemeMaster/files/11643325/es_launch_stdout.log)
[image]

JohnIrvine1433, 2023-06-04:
@SuperBadger77 the error is. It related to the changes, it seems that you're having an issue getting data from the OGA Gallery GitHub repo...
For the controllers I've tried another input file, can you try the latest version pushed on the x55 branch?

SuperBadger77, 2023-06-04:
@JohnIrvine1433 ok it now appears to work - I'm at the main menu (after checking for an update). However no controls work. Using my USB keyboard I can navigate but there is no screen size in settings matching the X55: 720×1280.

JohnIrvine1433, 2023-06-04:
@SuperBadger77 yeah do not update the app otherwise you'll get the master branch release. 
In gallery mode, ThemeMaster relies on Jetup's OGA Gallery information and at the moment there is no other resolution. Just choose the one best adapted to your device. 
If you know a GitHub repository with themes with the correct resolution I can add it to the GitHub mode. 
Regarding the controls, you have no key working at all?

SuperBadger77, 2023-06-04:
@JohnIrvine1433 I just retested it - no controls work - tried everything. Even Start + Select doesn't exit out. Had to use my keyboard to exit. If there are some files you need me to find on the x55 then let me know and I'll upload here. I'll try and install a few themes using the keyboard later on.

gannon345, 2023-06-05:
I just tested it and it was working, but then it prompted to update Thememaster and after it updated and relaunched, controls stopped working and it said the device was unsupported. It's the build previously provided: https://github.com/JohnIrvine1433/ThemeMaster/archive/refs/heads/x55.zip

So as long as you launch and don't run the update it works, though currently no themes support the Powkiddy's resolution (720p). I downloaded Artbook-3-2 though (IIRC it's 960x540) and it worked fine.

gannon345, 2023-06-05:
Okay, tested repeatedly, and yeah, the build at https://github.com/JohnIrvine1433/ThemeMaster/archive/refs/heads/x55.zip works. It's a little odd since it seems like B is confirm and A is back (reverse of the system controls) but otherwise seems fine as long as you don't try to update.

And as before, only thing lacking are 720p themes, but those probably aren't common. Even the default theme for the X55 is a 3:2 one so no big deal.

SuperBadger77, 2023-06-05:
@gannon345 I wonder why my controls don't work - did you change anything to get them to work?

gannon345, 2023-06-05:
@SuperBadger77 Did you update the app previously? I had to delete the folder and .sh script and copy them back in to get controls. It throws the "server issued error response" message, but on relaunch controls work, so long as you don't try to run the app update it prompts for on launch.

JohnIrvine1433, 2023-06-06:
@gannon345 i have pushed an updated tm-control file to the x55 branches that should fix the swap between A and B. 
@SuperBadger77 if you can try again with a clean install, it is also best to restart your device in case the old control file has not been properly closed... which may definitely happen if not handle properly. 
Do you still get the error message at first start?

gannon345, 2023-06-06:
@JohnIrvine1433 I ran the update process from Thememaster on my X55 and it went to 4.3.0 again. Did I need to manually install from the X55 branch zip? 

I'll test again via reinstall later today, won't be able to download it again for a few hours.

gannon345, 2023-06-07:
@JohnIrvine1433 Just tested, it's working like a champ.

JohnIrvine1433, 2023-06-09:
Perfect! I'll officially release a new version then. 
@gannon345 @SuperBadger77 thanks for you help

closed 2023-06-12 by JohnIrvine1433