JohnIrvine1433/ThemeMaster #15: ArkOS

Alpha-Inu, 2023-08-11:
ArkOS Latest release 07232023
RG353V Linux & Android
Panel Type 2/ID: 38 21

While using the latest version of ThemeMaster, selecting all themes in "Verified on (640x480)" and installing them en masse can cause ThemeMaster to crash back to the tools menu-BUT the confirmation button switched back to A from B, and selecting ThemeMaster from there opened Kodi. My attempts to back out of this menu were unsuccessful but all was fine once I rebooted.

So [I think] when it crashes this way, the tools menu is displayed on screen but Kodi Media Center is being activated from the main menu.

Notes

1. Installed Ark OS from v2.0_05042023.img, then updated to 07232023.

2. ThemeMaster was not present in the UI immediately after install, but the directory was present on the card. I've overwritten the files with the latest files from Github as referenced here:

#760 (comment)

3. I moved everything from SD2 and allowed Android to format the card, then I put everything back again-which seems to work fine for loading roms. I did this so that I could file manage with Android, which for some reason used to work fine but was telling me to format the card.

4. In ThemeMaster, 2 of the themes in "Verified on (640x480)" return a 404/html related error and replication might not work if they're selected.

"sagaworld by Jetup13"
"WOODY-CODY-RG351V by CodyV59"

JohnIrvine1433, 2023-08-12:
Hey, does it happen every time? I cannot reproduce and the two themes install correctly here... you might have an internet connection issue or something ?

Alpha-Inu, 2023-08-12:
I had to switch from a 5g network to a 2.4g network so I did at least 2 reboots and started ThemeMaster twice

First, I tried selecting <All> and was hit with the 404 error, the next time I individually selected all the themes starting with S and hit <Selected>. I believe those were the saga themes except sagaworld all the way down until WOODY-CODY.

It got through about 4 of them and then silently quit like an xbox

The only other thing I can think of to note is that VRAM limit is set to 150

I will look for the themes and delete them, then try again. Github was acting slow for me this week so maybe it had something to do with it

JohnIrvine1433, 2023-09-20:
So was the problem temporary ? Can I close the issue ?