JohnIrvine1433/ThemeMaster #16: Could we get support for the Powkiddy RGB30?

SuperBadger77, 2023-09-14:
My device just arrived - I'll try the current build and add it to my ports section to see if it works. Has a 720x720 resolution.

JohnIrvine1433, 2023-09-16:
@SuperBadger77 I do not have this platform, however I can implement the changes if you can provide me with the required information (input file for controls and event codes - you can get them by running evtest)

SuperBadger77, 2023-09-18:
I'm a bit of a numpty when it comes to this. If you let me know the steps to take I will gladly perform the task and share the details. I can ssh logon to my RGB30 running JELOS, do I just then run the command 'evtest' and paste the results here?

JohnIrvine1433, 2023-09-19:
Can you check whether you have this file : /dev/input/by-path/platform-gou_joypad-joystick
If yes I already got the key codes for it so you won't have to run evtest

SuperBadger77, 2023-09-19:
The only files I have in that folder are:
platform-adc-keys-event
platform-fdd40000.i2c-platform-rk805-pwrkey-event
platform-gpio-keys-event
platform-rk-headset-event
platform-singleadc-joypad-event-joystick
platform-singleadc-joypad-joystick

SuperBadger77, 2023-09-19:
If I need to run evtest - just let me know what you need me to do and I'll paste the results here.

JohnIrvine1433, 2023-09-20:
Ok, well, platform-singleadc-joypad-event-joystick file is already managed by the app, you should be able to run it without error. 
Do you have any issues with controls?

SuperBadger77, 2023-09-22:
I'll test it with the latest build and report back.

closed 2023-10-22 by JohnIrvine1433

SuperBadger77, 2023-11-21:
Seems to work fine.