JohnIrvine1433/ThemeMaster #18: ThemeMaster 4.4.2 and 4.5.0 Failing to Detect Connection on ArkOS v2.0 (10062023)

spirit-in-exile, 2023-10-07:
ThemeMaster starts, initializes, then displays:

>You appear to be offline. ThemeMaster require an internet access to run! [*sic*]

Connection verified and cycled, system restarted, no change.

Bug reported to christianhaitian/ArkOS here:
https://github.com/christianhaitian/arkos/issues/808#issue-1931166881

spirit-in-exile, 2023-10-07:
Update: ArkOS dev tested and it’s working, so it’s not ArkOS or ThemeMaster; must be something on my end. Network peculiarity, perhaps. Since it could not be replicated, closing. Thanks anyway.

spirit-in-exile, 2023-10-07:
Update: ArkOS dev tested and working fine, so it’s not ArkOS or ThemeMaster; perhaps network peculiarity or hardware issue on my end. Closing, thanks anyway.

closed 2023-10-07 by spirit-in-exile

retrotetro, 2024-08-07:
you needed to connect to wifi via android or usb port