JohnIrvine1433/ThemeMaster #20: ThemeMaster is nearly impossible to read on x55

milnak, 2024-01-22:
x55 uses a high dpi screen making this really hard to read.  Would be nice if it had a scalable UI like PortMaster does.

JohnIrvine1433, 2024-01-27:
Hi, this is definitely needed to be implemented then!
I'll try to do it in the following weeks but unfortunately I do not have an X55 device... would you be willing to beta test it for me?

JohnIrvine1433, 2024-01-28:
@milnak just made some adjustments to the menu dialog boxes in the new 4.6.0 prerelease. 
Could you check whether display is better now on x55?

JohnIrvine1433, 2024-02-18:
@milnak just checked latest release of PortMaster... they revamped UI moving away from bash dialog... this is really neat. It would be really nice to port it for ThemeMaster, I'll have a look at it but won't be able to work on it soon unfortunately.

JohnIrvine1433, 2025-07-19:
Full rewrite of UI would be required

closed 2025-07-19 by JohnIrvine1433