JohnIrvine1433/ThemeMaster #21: ThemeMaster crashes when starting on Ambernic 353V

Usertjeuh, 2024-02-05:
Installed the most recent version according to the manual.
It shows up under my Ports menu, but when i laungh it, the screens flickers 3 times and then returns to the ports menu. I assume it crashes for some reason.

JohnIrvine1433, 2024-02-07:
Seems to be linked to the environment, what's your OS? Have you tried to fully reboot your device after the installation ?

Usertjeuh, 2024-02-07:
Thanks for the reply! I'm running JelOS.

JohnIrvine1433, 2024-02-09:
Ok, I don't think there is something wrong on thememaster side unless of some under the hood changes in latest JELOS version. 
I'll try to reinstall latest JELOS release on my device asap to see if I can reproduce. I'll keep you updated.

JohnIrvine1433, 2024-02-18:
@Usertjeuh just reinstall JELOS, no problem here however I discovered that I couldn't get log from running main script. 
Can you install latest release (4.6.1) rerun ThemeMaster and see what message you're getting from /tmp/ThemeMaster.log ?

informadortech, 2024-03-13:
Hi,

I am using Jelos on a RG353VS, a RG353PS and a RGB30 and on neither thememaster is showing in ports or tools menu.

I unzip the thememaster to the roms/ports/ folder but it does not show up.

I am using a secondary sd card for the roms. Any help?

JohnIrvine1433, 2024-03-14:
ThemeMaster should appears in Ports menu once installed in /rom/ports/. 
ES restart can be needed for it to appear.

informadortech, 2024-03-14:
I've found the issue. Don't know why but gamelist.xml was not being updated. Just deleted the file and now it shows again. Sorry and thanks for your work!

closed 2024-03-15 by JohnIrvine1433