JohnIrvine1433/ThemeMaster #22: XU10 (Rk3326) compatibility progress

Andoramb, 2024-02-24:
Hi!

I've managed to get this running on RK3326 platform on a XU10 running Jelos, not sure if there was any progress on this, but here are some results:

ThemeMaster was not launching because no valid `param_device` was found. However, adding the following around [line 239](https://github.com/JohnIrvine1433/ThemeMaster/blob/8509d9ed3629f2afcf8aa39b75515d10e1ce0d00/ThemeMaster/ThemeMaster#L239)
```
  elif [[ -e "/dev/input/by-path/platform-xu10-joypad-event-joystick" ]]; then
    param_device="anbernic"
```
will launch the app.
Now navigation is still not possible, bc no valid inputs are found in `tm-joypad`
But through ssh it runs nicely: installing 640x480 themes is possible 👍

Andoramb, 2024-02-24:
[image]

But it looks navigation could be wrong, there should be more themes: for example I could `Elementerial` a month ago, now I'm unable to reach it.
(manually installing it works fine)

Thanks 🙂

JohnIrvine1433, 2024-02-24:
Hi!
Regarding input control, if you can run evtest against the input file that will give me to control codes I need to implement.

Andoramb, 2024-02-24:
I got this:
```
 evtest --grab /dev/input/event3
Input driver version is 1.0.1
Input device ID: bus 0x19 vendor 0xc3b0 product 0x200 version 0x10
Input device name: "xu10_joypad"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 304 (BTN_SOUTH)
    Event code 305 (BTN_EAST)
    Event code 307 (BTN_NORTH)
    Event code 308 (BTN_WEST)
    Event code 310 (BTN_TL)
    Event code 311 (BTN_TR)
    Event code 312 (BTN_TL2)
    Event code 313 (BTN_TR2)
    Event code 314 (BTN_SELECT)
    Event code 315 (BTN_START)
    Event code 316 (BTN_MODE)
    Event code 317 (BTN_THUMBL)
    Event code 318 (BTN_THUMBR)
    Event code 544 (BTN_DPAD_UP)
    Event code 545 (BTN_DPAD_DOWN)
    Event code 546 (BTN_DPAD_LEFT)
    Event code 547 (BTN_DPAD_RIGHT)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value      0
      Min    -1800
      Max     1800
      Fuzz      32
      Flat      32
    Event code 1 (ABS_Y)
      Value      0
      Min    -1800
      Max     1800
      Fuzz      32
      Flat      32
    Event code 3 (ABS_RX)
      Value      0
      Min    -1800
      Max     1800
      Fuzz      32
      Flat      32
    Event code 4 (ABS_RY)
      Value      0
      Min    -1800
      Max     1800
      Fuzz      32
      Flat      32
Properties:
Testing ... (interrupt to exit)
Event: time 1708775162.994958, type 1 (EV_KEY), code 544 (BTN_DPAD_UP), value 1
Event: time 1708775162.994958, -------------- SYN_REPORT ------------
Event: time 1708775163.151720, type 1 (EV_KEY), code 544 (BTN_DPAD_UP), value 0
Event: time 1708775163.151720, -------------- SYN_REPORT ------------
Event: time 1708775164.134951, type 1 (EV_KEY), code 545 (BTN_DPAD_DOWN), value 1
Event: time 1708775164.134951, -------------- SYN_REPORT ------------
Event: time 1708775164.285037, type 1 (EV_KEY), code 545 (BTN_DPAD_DOWN), value 0
Event: time 1708775164.285037, -------------- SYN_REPORT ------------
Event: time 1708775165.234977, type 1 (EV_KEY), code 546 (BTN_DPAD_LEFT), value 1
Event: time 1708775165.234977, -------------- SYN_REPORT ------------
Event: time 1708775165.368535, type 1 (EV_KEY), code 546 (BTN_DPAD_LEFT), value 0
Event: time 1708775165.368535, -------------- SYN_REPORT ------------
Event: time 1708775166.548301, type 1 (EV_KEY), code 547 (BTN_DPAD_RIGHT), value 1
Event: time 1708775166.548301, -------------- SYN_REPORT ------------
Event: time 1708775166.678382, type 1 (EV_KEY), code 547 (BTN_DPAD_RIGHT), value 0
Event: time 1708775166.678382, -------------- SYN_REPORT ------------
Event: time 1708775170.894930, type 1 (EV_KEY), code 305 (BTN_EAST), value 1
Event: time 1708775170.894930, -------------- SYN_REPORT ------------
Event: time 1708775170.988708, type 1 (EV_KEY), code 305 (BTN_EAST), value 0
Event: time 1708775170.988708, -------------- SYN_REPORT ------------
Event: time 1708775171.521700, type 1 (EV_KEY), code 307 (BTN_NORTH), value 1
Event: time 1708775171.521700, -------------- SYN_REPORT ------------
Event: time 1708775171.651583, type 1 (EV_KEY), code 307 (BTN_NORTH), value 0
Event: time 1708775171.651583, -------------- SYN_REPORT ------------
Event: time 1708775172.041539, type 1 (EV_KEY), code 304 (BTN_SOUTH), value 1
Event: time 1708775172.041539, -------------- SYN_REPORT ------------
Event: time 1708775172.141563, type 1 (EV_KEY), code 304 (BTN_SOUTH), value 0
Event: time 1708775172.141563, -------------- SYN_REPORT ------------
Event: time 1708775172.704990, type 1 (EV_KEY), code 308 (BTN_WEST), value 1
Event: time 1708775172.704990, -------------- SYN_REPORT ------------
Event: time 1708775172.861727, type 1 (EV_KEY), code 308 (BTN_WEST), value 0
Event: time 1708775172.861727, -------------- SYN_REPORT ------------
```
Do you need additional buttons?

JohnIrvine1433, 2024-02-24:
It looks like the event are similar to the RG552, can you try with the following code instead :

elif [[ -e "/dev/input/by-path/platform-xu10-joypad-event-joystick" ]]; then
    param_device="rg552"

JohnIrvine1433, 2024-02-24:
Ok thanks, I'll recompile the controls binary and will come back to you when it's done.

JohnIrvine1433, 2024-02-24:
Regarding themes listing, are you sure you miss some ? (the one you already installed no longer appears in the list)

Andoramb, 2024-02-24:
I 'm quite certain. I have recently reinstalled Jelos from scratch, and there are no other themes left to pick.
These ones are the first 6 Verified on the list (I have Albedo installed, and that is missing as expected)
but there are some more, however not possible to navigate to them.

Can it be connected to the implementation of `tput`?
Before that, on version 4.5.0 it was ok

JohnIrvine1433, 2024-02-25:
I don't think it's related to tput implementation which only changes height and width size of dialog boxes. 
May be there was a problem building the gallery, you can try rebuilding it by deleting everything in the data folder. 
From what I see in you screenshot the script didn't detect tput being installed and set default sizes for dialog boxes, as you've no scroll bar the cached gallery doesn't hold more theme...

Andoramb, 2024-02-25:
That's correct, tput is not found as a binary in Jelos

Andoramb, 2024-02-25:
On 4.6.2:
1. If I launch it through ssh: it works as it should 👍 
2. If I launch it via the Menu: at startup, it shows the new dialogbox: `"Controls can't be loaded, device not recognised. Please report for additional support."`
But that is shown for 1sec, then it starts normally.
*When quitting, it shows this new dialogbox again.

So it is possible to download and set themes this way :)

However navigating through the list is still a no-no

Andoramb, 2024-02-25:
Well I don't really know now...  I tried to revert to `4.5` release just to check the dependency on `tput`, but no difference vs `4.6.2`. So `tput` is not an issue 🙂 

Up/Down scroll is possible if there are more themes (for example in `GitHub repository` mode)
But with `Theme gallery by Jetup` I get only the first few themes:
[image]
And yeah, almost nothing is installed.

JohnIrvine1433, 2024-02-25:
Ok well there definitely is something wrong... may be there has been some changes on OGA Gallery from Jetup...
Could you set debug mode on and install latest prerelease and run it : https://github.com/JohnIrvine1433/ThemeMaster/archive/refs/tags/4.7.0.zip

Andoramb, 2024-02-25:
Sure, I just need to correct the previous comment:

"If I launch it via the Menu: at startup, ~~it shows the new dialogbox: "Controls can't be loaded, device not recognised. Please report for additional support."
But that is shown for 1sec, then it starts normally.
*When quitting, it shows this new dialogbox again.~~"
it works normally ✅ 
That dialogbox was a leftover from a previous check. Screen was not cleared. After a restart it's gone 🙂

Andoramb, 2024-02-25:
```
~/games-external/roms/ports/ThemeMaster/debug # cat debug_themes.txt 
************** NEW THEMES DEBUG OUTPUTS **************
Sun Feb 25 19:47:39 CET 2024
***** THEMEMASTER.CFG *****
app_conf_version=""
app_prerelease="off"
debug="on"
mode="B"
themes_location="/storage/.config/emulationstation/themes/"
temp_ram_folder="/dev/shm/"
app_autocheckupdate="on"
themes_autocheckupdate="off"
active_collection="Jetup13"
collections=(Jetup13 CodyV59 farfenkugell EmuELEC RetroPie)
app_colorscheme="All_Black"
gallery_user_res="640x480"
ttyoutput="/dev/tty"
nblines="15"
nbcols="50"
```
Jetup gallery:
[image]
GitHub repos list is OK, I can browse through 2O themes

JohnIrvine1433, 2024-02-25:
Please, can you list what you have under:
/storage/.config/emulationstation/themes
If you change the resolution, do you get the same results ?

Andoramb, 2024-02-26:
```
ls -lah /storage/.config/emulationstation/themes
total 10K
drwxr-xr-x    5 root     root        1.0K Feb 25 19:00 .
drwxr-xr-x    7 root     root        1.0K Feb 26 14:20 ..
drwxr-xr-x  172 root     root        5.0K Feb 25 19:00 es-theme-CODY-DARKTECK
drwxr-xr-x    5 root     root        1.0K Feb 24 10:05 es-theme-albedo
drwxr-xr-x    4 root     root        1.0K Feb  5 13:34 es-theme-elementerial
lrwxrwxrwx    1 root     root          40 Feb 23 19:38 system-theme -> /usr/share/themes/es-theme-art-book-next
```

JohnIrvine1433, 2024-02-26:
Ok so it all makes sense, in gallery mode data is cached locally and as the data file has been partially generated you missed some themes. 
The error is a bit strange as I already add the option not to check ssl certificate... may be there is something on the wget version in JELOS and/or igmur. 
I'll have a look into it. 
If we have the opportunity to run it again in debug mode, just delete everything in the data folder and once you ran it send me everything in the debug folder (in debug a save the wget output here).

Andoramb, 2024-02-27:
here's how it looks after a clean run:
```
RK3326:~/games-external/roms/ports/ThemeMaster # ls debug/
debug_themes.txt  tm_joypad.log
RK3326:~/games-external/roms/ports/ThemeMaster # cat debug/*
************** NEW THEMES DEBUG OUTPUTS **************
Tue Feb 27 14:04:38 CET 2024
***** THEMEMASTER.CFG *****
app_conf_version=""
app_prerelease="off"
debug="on"
mode="B"
themes_location="/storage/.config/emulationstation/themes/"
temp_ram_folder="/dev/shm/"
app_autocheckupdate="on"
themes_autocheckupdate="off"
active_collection="Jetup13"
collections=(Jetup13 CodyV59 farfenkugell EmuELEC RetroPie)
app_colorscheme="All_Black"
gallery_user_res="640x480"
ttyoutput="/dev/tty"
nblines="15"
nbcols="50"
RK3326:~/games-external/roms/ports/ThemeMaster # cat debug/tm_joypad.log 
--this file is empty--

RK3326:~/games-external/roms/ports/ThemeMaster # ls data/
es-theme-ARC-S-Theme-Remix---The-RA-CFW.png    es-theme-art-book-3-2.png                      es-theme-minimal-arkos.png
es-theme-CODY-DARKTECK.png                     es-theme-art-book-next-arkos.png               es-theme-nes-box.png
es-theme-CODY-DESTROY-THE-RG351V.png           es-theme-banner-ogsmax.png                     es-theme-pixel.png
es-theme-CODY-FLUX.png                         es-theme-base-text.png                         es-theme-sagaartbook.png
es-theme-CODY-FUCKYPLASTICS.png                es-theme-custom-box.png                        es-theme-sagabox.png
es-theme-ChicueloAP.png                        es-theme-elementerial.png                      es-theme-sagacrystal.png
es-theme-EPICMAX-CODY-RGB10MAX.png             es-theme-epic-cody-RG351P-M.png                es-theme-sagamodern.png
es-theme-RG351V-RetroBenny-NesBox-BlackBG.png  es-theme-epic-cody-RG351V.png                  es-theme-saganx.png
es-theme-RG351V-RetroBenny-NesBox-Wood.png     es-theme-epicnoir.png                          es-theme-sagaparagon.png
es-theme-RG351V-RetroBenny-PS2.png             es-theme-farfenkugell-columns.png              es-theme-sagaslice.png
es-theme-RG351V-RetroBenny-PS5.png             es-theme-farfenkugell_EPICMAX.png              es-theme-sagaworld.png
es-theme-RG3551V-RetroBenny-NesMini.png        es-theme-farfenkugell_crystal.png              es-theme-solo-horizontal-arkos-ed.png
es-theme-RetroNeon.png                         es-theme-farfenkugell_lazy.png                 es-theme-switch.png
es-theme-WOODY-CODY-RG351V.png                 es-theme-freeplay.png                          gallery.db
es-theme-albedo.png                            es-theme-gbz35_mod.png                         gallery.version
es-theme-arkos-carbon.png                      es-theme-magical-pixel.png
RK3326:~/games-external/roms/ports/ThemeMaster # cat data/gallery.db
*Account* *Repository* *480x320* *640x480* *854x480* *960x544*
mluizvitor es-theme-albedo Yes Yes Untested Untested
Jetup13 es-theme-arkos-carbon Yes Untested Untested Untested
Nacurutu es-theme-ARC-S-Theme-Remix---The-RA-CFW Untested Yes Untested Untested
Jetup13 es-theme-art-book-3-2 Yes Untested Untested Untested
nkahoang es-theme-art-book-next-arkos Untested Untested Untested Untested
Jetup13 es-theme-base-text Yes Yes Yes Yes
Jetup13 es-theme-banner-ogsmax No No Yes No
Jetup13 es-theme-ChicueloAP Yes Untested Untested Untested
CodyV59 es-theme-CODY-FLUX No Yes Untested Untested
CodyV59 es-theme-CODY-FUCKYPLASTICS No Yes Untested Untested
CodyV59 es-theme-CODY-DARKTECK No Yes Untested Untested
CodyV59 es-theme-CODY-DESTROY-THE-RG351V No Yes Untested Untested
Jetup13 es-theme-custom-box Yes No No No
mluizvitor es-theme-elementerial Yes Yes Untested Untested
Jetup13 es-theme-epicnoir Yes Yes Untested Untested
CodyV59 es-theme-epic-cody-RG351P-M Yes Untested Untested Untested
CodyV59 es-theme-epic-cody-RG351V Untested Yes Untested Untested
CodyV59 es-theme-EPICMAX-CODY-RGB10MAX Untested Untested Yes Untested
farfenkugell es-theme-farfenkugell-columns No No Yes Untested
farfenkugell es-theme-farfenkugell_crystal Yes Yes Yes Untested
farfenkugell es-theme-farfenkugell_EPICMAX Yes No No Untested
farfenkugell es-theme-farfenkugell_lazy Yes Yes Yes Untested
Jetup13 es-theme-freeplay Yes Yes Untested Untested
Jetup13 es-theme-gbz35_mod Yes Yes Untested Untested
Jetup13 es-theme-magical-pixel Yes Yes Untested Untested
Jetup13 es-theme-minimal-arkos Yes Yes Yes Yes
Jetup13 es-theme-nes-box Yes Yes Yes Untested
Jetup13 es-theme-pixel Yes Yes Yes Untested
TheBenny es-theme-RG351V-RetroBenny-NesBox-BlackBG No Yes No Untested
TheBenny es-theme-RG351V-RetroBenny-NesBox-Wood No Yes No Untested
TheBenny es-theme-RG3551V-RetroBenny-NesMini No Yes No Untested
TheBenny es-theme-RG351V-RetroBenny-PS2 Yes Yes No Untested
TheBenny es-theme-RG351V-RetroBenny-PS5 No Yes No Untested
Akira-N28 es-theme-RetroNeon Yes Yes Yes Yes
Jetup13 es-theme-sagaartbook Yes Yes Yes Yes
Jetup13 es-theme-sagabox Yes Yes Yes Yes
Jetup13 es-theme-sagacrystal Yes Yes Yes Yes
Jetup13 es-theme-sagamodern Yes Yes Yes Yes
Jetup13 es-theme-saganx Yes Yes Yes Yes
Jetup13 es-theme-sagaparagon Yes Yes Yes Yes
Jetup13 es-theme-sagaslice Yes Yes Yes Yes
Jetup13 es-theme-sagaworld Yes Yes Yes Yes
Jetup13 es-theme-solo-horizontal-arkos-ed Yes Untested Untested Yes
Jetup13 es-theme-switch Yes Yes Yes Yes
CodyV59 es-theme-WOODY-CODY-RG351V No Yes Untested Yes
RK3326:~/games-external/roms/ports/ThemeMaster # cat data/gallery.version
2024-01-29T19:58:12Z
```
This time everything was downloaded OK, the list was fully populated 👍 

Btw what is the reason to download png-s? 🙂

JohnIrvine1433, 2024-02-27:
Ok, I had a look in my side and it seems that gnutls may have had a problem which the ssl handshake, could be because of igmur site or problem on whet side. 
As it seems to be temporary I'll implement a retry to ensure that the gallery database is fully populated (I'll do it for the final release of 4.7.0)
I'm downloading the png as part of gallery data in order to display themes preview when you browse the gallery.

Andoramb, 2024-02-28:
Right.
I haven't seen a device to use the preview feature (yet)
But would it make sense to skip png-s altogether in some devices or based on some conditions like `tput`?
Like if the resolution is 640x480 or less + everything runs in a terminal + it gives ssl errors .. 🤷‍♂️

Thanks for your support! 🙂

JohnIrvine1433, 2024-03-01:
When you're browsing themes, if you select "Details" button you'll see the preview picture.

Andoramb, 2024-03-01:
...not on XU, that's fo' sure :D
`Details` won't show anything for selected/unselected theme. Not on the device nor in ssh :)

Andoramb, 2024-03-01:
Sorry, one more thing for the XU10, just to have a complete picture:
I've seen in [ThemeMaster line 337](https://github.com/JohnIrvine1433/ThemeMaster/blob/513d0c25700b49befb1a09718b24dc2c6a8ffce7/ThemeMaster/ThemeMaster#L337) that it looks for JELOS and UnofficialOS.
There is a new kid around the block, that looks to be based on JELOS, it's called `NOSTIX', from
```
cat /etc/os-release
OS_NAME="NOSTIX"
```
So it's essentially the same.
Could you add this OS to the condition line above as well? 🙂

JohnIrvine1433, 2024-03-01:
Ok no problem I'll add support for NOSTIX. 
Regarding the Details button, you should get a picture preview if it's been cached (png available) and otherwise a text description of the theme extracted from GitHub.  
Don't you get anything at all?

JohnIrvine1433, 2024-03-03:
Ok I'll close the issue since xu10 feature has been implemented. 
If you still have nothing when selecting Details for a theme in Gallery mode please open another issue and we'll dig into it.

closed 2024-03-03 by JohnIrvine1433

JohnIrvine1433, 2024-03-04:
@Andoramb today i'm getting unknown error 429 from imgur ... impossible to get anything from wget or curl... while it's working fine from a browser

Andoramb, 2024-03-04:
hey, I'll check this and the previous things tomorrow before lunch :)
Got Discord btw?

JohnIrvine1433, 2024-03-05:
Yeah, @johnirvine

JohnIrvine1433, 2024-03-05:
@Andoramb you should no longer have the problem, pictures have been moved from igmur to GitHub