JohnIrvine1433/ThemeMaster #26: Names Platforms

aksenav2, 2024-04-11:
Hello, please tell me where are located in the arkos system, the names of the platforms/emulators. I didn't find them in the topic resources. 
[image]

JohnIrvine1433, 2024-04-11:
Hi, I think what you're referring to is part of ES theme, text can be defined in theme.xml file or part of the image.

aksenav2, 2024-04-12:
> Hi, I think what you're referring to is part of ES theme, text can be defined in theme.xml file or part of the image.

There are no such texts in the resources of the themes.

JohnIrvine1433, 2024-04-12:
What theme are you using ? It might be part of the picture...

aksenav2, 2024-04-12:
> What theme are you using ? It might be part of the picture...
ES-THEME-ALBEDO No, it's not part of the graphics, I've redrawn almost all of the graphics to my own.

JohnIrvine1433, 2024-04-12:
Don't know about it, best would probably to ask mluizvitor

aksenav2, 2024-04-12:
> Don't know about it, best would probably to ask mluizvitor

Thank you

closed 2024-04-12 by aksenav2