JohnIrvine1433/ThemeMaster #27: Thememaster crashes upon selecting "Uninstall a theme"

Giuleus957, 2024-05-01:
I'm running latest ArkOS (04/24/2024) and latest Thememaster (4.8.1) on my Anbernic RG353VS and when I select "Uninstall a theme" Thememaster crashes. All the rest of Thememaster works fine, I can install new themes etc. but crashes only when I select "Uninstall a theme".

manasv, 2024-05-04:
Same for me I think is something related to the latest update (4.8.1) as I reverted to (4.8.0) and it stopped crashing but still I have some issues because is not able to fetch data from the repositories because of SSL failure.

JohnIrvine1433, 2024-05-18:
Hi everyone, sorry for the late reply I was away for a while. I'll fix it in the coming day.

JohnIrvine1433, 2024-05-20:
I have pushed v4.9.0 as prerelease, should do the job.

manasv, 2024-05-21:
Can confirm that 4.9.0 fixed the problem on my side, thank you.

closed 2024-05-21 by JohnIrvine1433