JohnIrvine1433/ThemeMaster #29: Support For Anbernic RG-ARC

SuperBadger77, 2024-05-21:
Hi,

RockNix recently introduced support for the RG-ARC. Would it be possible to get Theme master support for it? I just tried the latest release and all I get is a blank screen. The controls are different but the chipset and screen is the same as the RG503.

JohnIrvine1433, 2024-05-21:
Hi, I can definitely work on it, as I do not have the device if you can give me the input file used (should be under /dev/input/by-path/) and run evtest against it so I can get the key codes to implement that would be great. 

I'm also working on ROCKNIX support. I already fixed the execution issue which gave the blank screen but it seems that I have to change the way dialog boxes are managed to have them correctly displayed (I don't really know why but bash doesn't handle them properly on ROCKNIX).

SuperBadger77, 2024-05-21:
RG-ARC-D:/dev/input # cd by-path/
RG-ARC-D:/dev/input/by-path # ls
platform-adc-keys-event
platform-fdd40000.i2c-platform-rk805-pwrkey.3.auto-event
platform-fe0a0000.hdmi-event
platform-fe5b0000.i2c-event
platform-gpio-keys-vol-event
platform-pwm-vibrator-event
platform-rg_arc_joypad-event-joystick
platform-rg_arc_joypad-joystick
platform-sound-event

SuperBadger77, 2024-05-21:
Which one do I need? I can't access the sd card on Windows via Linux File System by Paragon - for some reason it doesn't work on Jelos / Rocknix builds - something they did to the file system a while back.
Do you need the file? Or do you just need me to run evtest against a particular file?

SuperBadger77, 2024-05-21:
I ran evtest - then is asked me to choose so I chose event 6 rg-arc-joypad:
# evtest
No device specified, trying to scan all of /dev/input/event*
Available devices:
/dev/input/event0:      pwm-vibrator
/dev/input/event1:      rk805 pwrkey
/dev/input/event2:      dw_hdmi
/dev/input/event3:      adc-keys
/dev/input/event4:      rk817_ext Headphones
/dev/input/event5:      gpio-keys-vol
/dev/input/event6:      rg_arc_joypad
/dev/input/event7:      Goodix Capacitive TouchScreen
Select the device event number [0-7]: 6
Input driver version is 1.0.1
Input device ID: bus 0x19 vendor 0x1 product 0x1 version 0x100
Input device name: "rg_arc_joypad"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 304 (BTN_SOUTH)
    Event code 305 (BTN_EAST)
    Event code 306 (BTN_C)
    Event code 307 (BTN_NORTH)
    Event code 308 (BTN_WEST)
    Event code 309 (BTN_Z)
    Event code 310 (BTN_TL)
    Event code 311 (BTN_TR)
    Event code 312 (BTN_TL2)
    Event code 313 (BTN_TR2)
    Event code 314 (BTN_SELECT)
    Event code 315 (BTN_START)
    Event code 544 (BTN_DPAD_UP)
    Event code 545 (BTN_DPAD_DOWN)
    Event code 546 (BTN_DPAD_LEFT)
    Event code 547 (BTN_DPAD_RIGHT)
Properties:
Testing ... (interrupt to exit)
Event: time 1716323499.507816, type 1 (EV_KEY), code 547 (BTN_DPAD_RIGHT), value 1
Event: time 1716323499.507816, -------------- SYN_REPORT ------------
Event: time 1716323499.600279, type 1 (EV_KEY), code 547 (BTN_DPAD_RIGHT), value 0
Event: time 1716323499.600279, -------------- SYN_REPORT ------------
Event: time 1716323500.322060, type 1 (EV_KEY), code 546 (BTN_DPAD_LEFT), value 1
Event: time 1716323500.322060, -------------- SYN_REPORT ------------
Event: time 1716323500.386787, type 1 (EV_KEY), code 546 (BTN_DPAD_LEFT), value 0
Event: time 1716323500.386787, -------------- SYN_REPORT ------------
And then I did a left and right on the joypad at the end.

JohnIrvine1433, 2024-05-23:
@SuperBadger77 here is a pre-build supporting RG-ARC and ROCKNIX ; I still have a problem with the image viewer not running on ROCKNIX but I can't figure out why at the moment... otherwise everything else should be ok
[ThemeMaster-nightly.zip](https://github.com/JohnIrvine1433/ThemeMaster/files/15421315/ThemeMaster-nightly.zip)

SuperBadger77, 2024-05-24:
Will test this weekend and report back.

SuperBadger77, 2024-05-24:
All works ok! Can see the screen, downloaded a couple of themes. Applied ok. Ship it!

closed 2024-05-25 by JohnIrvine1433