JohnIrvine1433/ThemeMaster #30: ThemeMaster isn´t working in Arkos 05242024 version

Alvxr0Lopz, 2024-05-26:
I just updated today the OS in my R36S and also updated PortMaster and ThemeMaster to v4.10, everything is fine with portmaster, but when I try to open ThemeMaster it starts for a few seconds and suddenly closes

christianhaitian, 2024-05-26:
Does the same on the rg351v.  I was trying to diagnose the issue but it launches fine from ssh.  Not sure what the issue is just yet.

christianhaitian, 2024-05-26:
Version 4.9 works just fine.  So it's something with a change for 4.10.

Aezriel82, 2024-05-27:
just moved over from jelos to arkos on rgb30 never used previous versions but 4.10 is same, starts fine as soon as it gets to the UI is closes. 
where can i get the 4.9 to test?

christianhaitian, 2024-05-27:
Here: https://github.com/JohnIrvine1433/ThemeMaster/archive/672d7456c22e1f00b74762f63461727c5b162e06.zip
Just don't update it until the maintainer has had a chance to review and address the issue.

JohnIrvine1433, 2024-05-27:
I'll be on it shortly, problem is coming from change relying on $(tty) to properly display dialog boxes depending on environnement but on ArkOS this return "not a tty" (and I you can guess I've not tested the modification on ArkOS... too many OSs out now for me to go through them all...)

JohnIrvine1433, 2024-05-28:
All, v4.10.1 has just been released fixing the issue on ArkOS

fabinhojcr, 2024-05-29:
Now when trying to enter Thememaster an error message appears saying that I'm not connected to the Internet on the r36s even though I'm connected to Wi-Fi.

JohnIrvine1433, 2024-05-29:
@fabinhojcr nothing has ever change on that side, the message means that the script couldn't successfully ping google.com
Have you been able to check internet connectivity outside of ThemeMaster?

fabinhojcr, 2024-05-29:
@JohnIrvine1433 Yes, everything connects to the Internet, only the Thememaster doesn't.

closed 2024-05-29 by JohnIrvine1433