JohnIrvine1433/ThemeMaster #31: RGB30 Not working with Rocknix

aenertia, 2024-06-13:
Using latest HEAD and/or the release.

Launching results in a 'unable to find control' dialog. Adding the correct /dev/<event> path for the Mainline joypad and setting to x55/anbernic results in non-usable controls but the app launching into menu.

Rocknix has Sway/wl compositor - so should thememaster simply inherit/check and inherit the wayland input controls?

JohnIrvine1433, 2024-06-13:
Could you confirm the /dev/input/ you're having for pad's control and run evtest against it?

josemalop, 2024-06-29:
Same issue on Anbernic RG353P.

JohnIrvine1433, 2024-06-29:
@josemalop can you try and report with the enclosed file ?
[ThemeMaster.zip](https://github.com/user-attachments/files/16041403/ThemeMaster.zip)

JohnIrvine1433, 2024-06-29:
@josemalop could you try putting the enclosed file in ThemeMaster's folder and report if it's working ?
[paramcontrols.txt](https://github.com/user-attachments/files/16041506/paramcontrols.txt)

JohnIrvine1433, 2024-06-29:
I have published v4.11.1 as pre-release, from everything I gather it should fix the issue but I cannot confirm as I do not have a device with an rk3588 chip ... Could one of you test it ? @aenertia @josemalop

josemalop, 2024-07-01:
Works like a charm!! Thank you very much for your effort! :-)

closed 2024-07-02 by JohnIrvine1433