JohnIrvine1433/ThemeMaster #32: Controls can't be loaded, device not recognized.

JosephLGuernsey, 2024-06-23:
I download and followed the install instructions for Rocknix. when i launch from ports menu it look to be loading then says, "Controls can't be loaded, device not recognized. Please report for additional support."
I am using RBG30 with Rocknix OS (ROCKNIX-RK3566.aarch64-20240612.img)
Is there a way to fix this?
[image]

JohnIrvine1433, 2024-06-23:
Could you confirm the /dev/input/ you're having for pad's control and run evtest against it?

JosephLGuernsey, 2024-06-23:
Sorry I'm not sure how to do that. I did find the /dev/input/ folder. It has some files and a folder called by-path in it.

JohnIrvine1433, 2024-06-23:
I think I know why it's not working from another issue reported. 
- I need to check how to properly fix it in the long run, in the mean time you can try to bypass it by using manual settings by creating a "paramcontrols.txt" file within ThemeMaster's folder with the following content:
`back_key = 314;
start_key = 315;
a_key = 305;
b_key = 304;
x_key = 307;
y_key = 308;
up_key = 544;
down_key = 545;
left_key = 546;
right_key = 547;
l1_key = 310;
l2_key = 312;
r1_key = 311;
r2_key = 313;
inputstr = /dev/input/by-path/platform-rocknix-singleadc-joypad-event-joystick;`

JosephLGuernsey, 2024-06-24:
The "paramcontrols.txt" file did bypass the issue, It got me to the menu. The controls on the handheld would not work. I was able to get through the menus with a Bluetooth mouse. Happy to report I have themes now. Thank you for your help and awesome app!

closed 2024-06-24 by JosephLGuernsey

JohnIrvine1433, 2024-06-24:
@JosephLGuernsey Could you confirm the list of input files you have under /dev/input/by-path/ ?

JosephLGuernsey, 2024-06-25:
Yes here you go, let me know if you need anything else.

/dev/input/by-path/
[image]

/dev/input/
[image]

JohnIrvine1433, 2024-06-25:
Thanks, they've changed input file name on latest ROCKNIX release which is why ThemeMaster is no longer recognising it. However based on the device you're using and the input file name I don't understand why you don't have controls when using the paramcontrols file...