JohnIrvine1433/ThemeMaster #33: Doesn't Detect Controller Type On X55 with Newest Rocknix Release 20240702

dewbieZ, 2024-07-05:


JohnIrvine1433, 2024-07-06:
They must have changed the input driver and associated input file name. 
Could you list me what you're having in /dev/input/by-path/ ?

dewbieZ, 2024-07-08:
RK3566-X55:/dev/input # cd by-path
RK3566-X55:/dev/input/by-path # ls
platform-fdd40000.i2c-platform-rk805-pwrkey.2.auto-event
platform-fe0a0000.hdmi-event
platform-gpio-keys-vol-event
platform-rocknix-joypad-event-joystick
platform-rocknix-joypad-joystick
platform-sound-event
RK3566-X55:/dev/input/by-path #

dewbieZ, 2024-07-08:
I get a segmentation fault and crash after changing in main ThemeMaster shell script with tm-joypad program

JohnIrvine1433, 2024-07-09:
@dewbieZ give it a try using manual parameters by putting the enclosed file in the ThemeMaster folder. 
[paramcontrols.txt](https://github.com/user-attachments/files/16149801/paramcontrols.txt)

tc8510, 2024-08-14:
Adding the .txt worked! Thank you!

brupec, 2024-09-21:
worked for me too thanks!