JohnIrvine1433/ThemeMaster #36: Issue

lauren7ino, 2024-08-15:
Hi, tonight I thought it would be great to change my theme.
I have and rg353m running with ArkOS latest.
First off, I noticed that I could not see previews.
Then, I tried installing all themes for 640 x 480, but some 404 error appeared.
Then, I went to settings and swapped the settings to use GitHub.
Then I installed all the themes at once, using the special button.
All went well (?), but then when I closed and reopened ThemeMaster, it checks for updates on all the installed Themes, gives 404 error, and asked me to exit.
So, I can no longer use theme master...
Help?

JohnIrvine1433, 2024-08-16:
Can you set ThemeMaster in debug mode, delete the "data" folder and run it again in order to get the issues. 
Once done send me the content of the debug folder that would have been created.

lauren7ino, 2024-08-17:
Hi
Sorry, the 404 mysteriously no longer happens today, so I can no longer test the issue.
It seems to me as a developer, in case of error, ThemeMaster should skip all auto-updates, instead of exiting.
Basically turn these off:
app_autocheckupdate="on"
themes_autocheckupdate="on"

If the situation repeats, I now know how to turn on debug mode.
Anyway, here's my configuration:

app_conf_version="4.10.0"
app_prerelease="off"
debug="off"
mode="A"
themes_location="/roms/themes/"
temp_ram_folder="/dev/shm/"
app_autocheckupdate="on"
themes_autocheckupdate="on"
active_collection="Jetup13"
collections=(Jetup13 CodyV59 farfenkugell EmuELEC RetroPie)
app_colorscheme="All_Black"
gallery_user_res="640x480"
nblines="14"
nbcols="48"

Thanks

lauren7ino, 2024-08-17:
It seems I managed to reproduce the problem when I swapped the Themes repository.
The current ThemeMaster behavior makes this problem unrecoverable for most non technical users.

https://github.com/user-attachments/assets/27d9aae6-7259-4ad6-a18e-ed7b13d06883

JohnIrvine1433, 2024-08-17:
I'll change the app behaviour so it starts even if it encounters retrieving errors at startup. 
Regarding your 403 error message this is expected once you've reached the maximum threshold allowed by GitHub for anonymous requests.

lauren7ino, 2024-08-17:
Thanks. I'll be looking for it.

lauren7ino, 2024-08-17:
Hi, I just installed all themes from all repositories, because before installing preview did nothing.
Now, I find out that I cannot list or preview any of the installed themes.
What am I missing?

JohnIrvine1433, 2024-08-18:
> Hi, I just installed all themes from all repositories, because before installing, preview did show nothing.
> Now, I find out that I cannot list or preview any of the installed themes.
> What am I missing?

Once themes are installed you can list and try them directly via ES UI menu. 
There is no need to list or preview them in ThemeMaster (but you can list installed themes via the Uninstall menu).

lauren7ino, 2024-08-18:
The problem with "try them directly via ES UI menu" is that once a bad theme fails to load, ES is no longer usable.
I've already found at least 2 of these.

lauren7ino, 2024-08-18:
I see no Preview option anywhere in ThemeMaster.

lauren7ino, 2024-08-18:
By the way, are you protecting against 403 / connection loss while downloading the themes?
I suspect the few bad themes I have downloaded are incomplete because of 403 errors during the download.
If you are not doing this, download the files to a temporary folder name, and only rename to the final folder name if the theme was completely downloaded, otherwise delete the temporary folder.

JohnIrvine1433, 2024-08-18:
> The problem with "try them directly via ES UI menu" is that once a bad theme fails to load, ES is no longer usable.
> 
> I've already found at least 2 of these.

What do you mean by "bad theme fails to load"?
Themes are directly downloaded from GitHub as a zip file. Corrupted downloads won't be installed. If a theme is not properly working once installed it might be a problem with the theme itself (however the ones from the Gallery mode have been checked and should work just fine).

JohnIrvine1433, 2024-08-18:
> I see no Preview option anywhere in ThemeMaster.

Previews are available in Gallery mode from the Browse menu, when you select a theme you'll have access to the Preview option.

JohnIrvine1433, 2024-08-18:
> By the way, are you protecting against 403 / connection loss while downloading the themes?
>
> I suspect the few bad themes I have downloaded are incomplete because of 403 errors during the download. [...]

This is already dealt with.

lauren7ino, 2024-08-18:
This preview?

https://github.com/user-attachments/assets/39550d58-2534-42ad-97d7-a91e1fe02458

lauren7ino, 2024-08-18:
> What do you mean by "bad theme fails to load"? Themes are directly downloaded from GitHub as a zip file. Corrupted downloads won't be installed. If a theme is not properly working once installed it [...]

Since they are gzipped, it's probably a badly created theme.
I will do further testing later.

JohnIrvine1433, 2024-08-18:
> This preview?
> 
> 
> 
> https://github.com/user-attachments/assets/39550d58-2534-42ad-97d7-a91e1fe02458
> 
> 
> 
> 

Yeah, not normal, don't have the issue on my side, might be an issue linked to your OS version. 
If you ran it in debug mode you may have some details about the issue in the image_viewer.log file within the debug folder. 
Alternatively you can try to directly run the image viewer with the path to an png file as parameter.

JohnIrvine1433, 2024-10-11:
All fixed in the latest release

closed 2024-10-11 by JohnIrvine1433