JohnIrvine1433/ThemeMaster #37: Font size/screen real-estate

lauren7ino, 2024-08-18:
rg353m: 640x480 screen

1) Is there any hack I can edit in the config file to double the font size? It's barely readable.
2) What about to improve the wasted screen space?

Thanks
[image]

JohnIrvine1433, 2024-08-18:
Because of the wide variety of handheld devices out there it is hard to get a proper setting for each of them. 
The script tries to relies in tput, stty size and $LINES variable to adjust messages boxes, a default setting is set for smaller screen available of voice of the environment test can get a setting for the number of lines and columns for the terminal... which is what you end up with. 
What I'll do is adding a custom variable that can be changed within the configuration file so people can adjust the default setting.

JohnIrvine1433, 2024-08-26:
@lauren7ino you should be able to manually change the configuration editing the ThemeMaster.cfg file (nbcols and nblines variables)

lauren7ino, 2024-08-26:
Thanks but it doesn't detect a new version yet. Logs report v4.11.1

JohnIrvine1433, 2024-08-27:
> Thanks but it doesn't detect a new version yet. Logs report v4.11.1

Should already be available in 4.11.1 (couldn't remembrer I'd already set the variables)

JohnIrvine1433, 2024-08-27:
> Q: If you already have a menu with resolutions, why not use these resolutions to change the number of rows/cols ?

I'll have a look into it, I need to check how I can get current terminal font size in order to do it.

JohnIrvine1433, 2024-10-11:
@lauren7ino Version 5.0.0 should better manage it and you can always change it in ThemeMaster.cfg if needed (can't change font size though so might not be perfect)

closed 2024-10-11 by JohnIrvine1433

lauren7ino, 2024-10-11:
The screen now is used in its full height, unfortunately at 720p the fonts are as small as flees.
I can't use your app without a magnifying glass.
And I'm not kidding!
(I'm 52 btw)

lauren7ino, 2024-10-11:
Thank you for your effort. 
I understand your struggle. Unfortunately I have no Linux experience to help.
Thank you again.