JohnIrvine1433/ThemeMaster #38: Bug: Browse Gallery fails to launch preview or download themes if any theme repo fails

skomae, 2024-09-06:
**Outcome:**
When selecting _Browse Gallery_ -> _Themes tested on 640x480 displays_ : -> (select some repos) -> _Preview_, ThemeMaster fails with the following message.

Server issued an error response.
https://api.github.com/repos//0/commits:
2024-09-06 18:56:38 ERROR 404: Not Found

And offers to quit or continue back to the main menu. 

**Expected Outcome:**
ThemeMaster either should continue to attempt downloading the next theme, or allow failed repos (likely closed, or inaccessible) to fail silently and present the user with a list of failed theme downloads.

**Steps to reproduce:**
Attempt to download all themes from the gallery which must contain an unreachable repo to reproduce.

JohnIrvine1433, 2024-09-07:
I'll add this feature to my todo list!

However it seems that you're getting an error because of improperly set variables being used to build the URL (GitHub user is empty and the repository name is 0 : https://api.github.com/repos//0/commits). 

Could you activate the debug mode in settings menu and reproduce the error?
Then send me the content of the debug folder that would have been created in the app folder.

skomae, 2024-09-07:
Thanks for the quick reply @JohnIrvine1433

Here are the logs: https://gist.github.com/skomae/e31e816588b56699ee45e7884265d8ce

In order to get the logs to write, I had to `mkdir debug` and `touch debug/debug_themes` in the ThemeMaster directory.

Running ThemeMaster 4.11.1 on ArkOS 2.0 08232024 RG353V. There could be an issue with the stock configuration provided with the OS. 

However it would still be extremely helpful if ThemeMaster could skip over repos it can't reach!

JohnIrvine1433, 2024-10-11:
Implemented in 5.0.0 (batch install error no longer ends installation process)

closed 2024-10-11 by JohnIrvine1433