JohnIrvine1433/ThemeMaster #40: R36S problems installing updates

LobAnovich, 2024-10-22:
R36S (not a clone), O30S (horizontal version of R36S), Theme master updated to version 5.0.0. Finds theme update, downloads, installs, but when re-checking it again finds updates for these themes. So it can't install theme updates. The last time I launched it found an update for Theme Master itself to 5.1.0, it pretends to update, writes a change log, but in fact no update occurs. When I launch it again, the theme master version is 5.0.0. It again finds update 5.1.0. The same thing with themes. On one device, arkos from 09.29.24, on the other, arkos from AeolusUX from 08.23.24-1. The same behavior on both devices

JohnIrvine1433, 2024-10-23:
Should be fixed in 5.1.1

closed 2024-10-23 by JohnIrvine1433

LobAnovich, 2024-10-27:
With the new 5.1.1 fix, the main part of the problem disappeared, but the problem with constantly finding an update to the "switch theme", even after installing this update, remained