JohnIrvine1433/ThemeMaster #41: Retroid Pocket 5 Support

SuperBadger77, 2024-12-18:
Can I request support for Retroid Pocket 5 on RockNix? Happy to do any required testing

JohnIrvine1433, 2024-12-18:
Hi, we can definitely work on it. (we probably gonna have to also do some tweaks for the UI as the device resolution is quite big compared to to the other ones...)
Could you start by telling me what you're having in /dev/input/by-path/ ?

SuperBadger77, 2024-12-19:
platform-888000.serial-event-mouse
platform-a94000.i2c-event
platform-c440000.spmi-platform-c440000.spmi:pmic@0:pon@800:pwrkey-event
platform-c440000.spmi-platform-c440000.spmi:pmic@0:pon@800:resin-event
platform-gpio-keys-event
platform-sound-event

JohnIrvine1433, 2024-12-19:
Ok, not sure about the right input file to be used, can you try tuning evtest against platform-gpio-keys-event may be?
I need to code event for each keypad and joypad direction.

SuperBadger77, 2024-12-19:
I am technical but a noob at linux. If you give me exact instructions I'll come back with the results. Tell me what to run, assume at the cmd line prompt and I'll press all the buttons and paste the results here.

JohnIrvine1433, 2024-12-19:
Just connect to ROCKNIX and run : evtest /dev/input/by-path/platform-gpio-keys-event
You should get default configuration output and then you can get the different event codes by pressing each button. 
(You may need to run it with sudo)

Tom2112, 2025-01-06:
Just want to add that I would love to see support for the Retroid Pocket 5 in Thememaster!

SuperBadger77, 2025-01-06:
I am back off holiday - will try and get the data to you this wee. @Tom2112 if you have a RP5 and can run the above test that would help get things moving quicker :-)

Tom2112, 2025-01-06:
I don't think this looks like enough events (all the buttons/directions) but this is what I got from my RP5:
`login as: root
root@192.168.4.154's password:

███████   ████████  ████████  ██   ██  ████    ██  ████████  ██    ██
██    ██  ██    ██  ██    ██  ██  ██   ██ ██   ██     ██      ██  ██
██████    ██    ██  ██        █████    ██  ██  ██     ██       ████
██   ██   ██    ██  ██    ██  ██  ██   ██   ██ ██     ██      ██  ██
██    ██  ████████  ████████  ██   ██  ██    ████  ████████  ██    ██
... Version: 20241120-official (cce0e1d)
... Built: Wed Nov 20 02:41:05 UTC 2024

SD865:~ # evtest /dev/input/by-path/platform-gpio-keys-event
Input driver version is 1.0.1
Input device ID: bus 0x19 vendor 0x1 product 0x1 version 0x100
Input device name: "gpio-keys"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 115 (KEY_VOLUMEUP)
Properties:
Testing ... (interrupt to exit)
`

Tom2112, 2025-01-06:
This is what's in my /dev/input/by-path/

`SD865:~ # ls -al /dev/input/by-path/
total 0
drwxr-xr-x    2 root     root           160 Jun 25  2024 .
drwxr-xr-x    3 root     root           180 Jun 25  2024 ..
lrwxrwxrwx    1 root     root             9 Jun 25  2024 platform-888000.serial-event-mouse -> ../event2
lrwxrwxrwx    1 root     root             9 Jun 25  2024 platform-a94000.i2c-event -> ../event3
lrwxrwxrwx    1 root     root             9 Jun 25  2024 platform-c440000.spmi-platform-c440000.spmi:pmic@0:pon@800:pwrkey-event -> ../event0
lrwxrwxrwx    1 root     root             9 Jun 25  2024 platform-c440000.spmi-platform-c440000.spmi:pmic@0:pon@800:resin-event -> ../event1
lrwxrwxrwx    1 root     root             9 Jun 25  2024 platform-gpio-keys-event -> ../event5
lrwxrwxrwx    1 root     root             9 Jun 25  2024 platform-sound-event -> ../event4
`

Tom2112, 2025-01-06:
If you're feeling froggy, here's the output of that command for the RP5 as well as a few other hires devices. The other devices didn't have /dev/input/by-path/platform-gpio-keys-event, so I showed the output for the events that were there - probably more info than you wanted, hopefully including what you needed!

[Powkiddy RGB20 Pro.txt](https://github.com/user-attachments/files/18325604/Powkiddy.RGB20.Pro.txt)
[Retroid Pocket 5.txt](https://github.com/user-attachments/files/18325605/Retroid.Pocket.5.txt)
[Powkiddy X55.txt](https://github.com/user-attachments/files/18325606/Powkiddy.X55.txt)
[Powkiddy RGB10 MAX3.txt](https://github.com/user-attachments/files/18325607/Powkiddy.RGB10.MAX3.txt)

SuperBadger77, 2025-01-07:
@JohnIrvine1433 do you have enough to go on there or do you need more data?

JohnIrvine1433, 2025-01-08:
@Tom2112 @SuperBadger77 
Usually there is a file with "joypad-event" in its name or something alike which handles the joypad keys.
As it is not the "platform-gpio-keys-event" one, can you try with the other ones ?
The evtest should output the list of buttons like BTN_SOUTH, BTN_EAST, etc.
When you get the right output you can confirm that it is the correct file by pressing each button and direction key (this should output the event code). thanks.

Tom2112, 2025-01-09:
I think I found the one you're looking for. I included me hitting all of the buttons/axis/sticks:

`RetroidPocket5:~ # evtest /dev/input/by-path/platform-888000.serial-event-mouse
Input driver version is 1.0.1
Input device ID: bus 0x3 vendor 0x2020 product 0x3001 version 0x1
Input device name: "Retroid Pocket Gamepad"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 278 (BTN_BACK)
    Event code 304 (BTN_SOUTH)
    Event code 305 (BTN_EAST)
    Event code 307 (BTN_NORTH)
    Event code 308 (BTN_WEST)
    Event code 310 (BTN_TL)
    Event code 311 (BTN_TR)
    Event code 314 (BTN_SELECT)
    Event code 315 (BTN_START)
    Event code 316 (BTN_MODE)
    Event code 317 (BTN_THUMBL)
    Event code 318 (BTN_THUMBR)
    Event code 544 (BTN_DPAD_UP)
    Event code 545 (BTN_DPAD_DOWN)
    Event code 546 (BTN_DPAD_LEFT)
    Event code 547 (BTN_DPAD_RIGHT)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value     72
      Min    -1408
      Max     1408
    Event code 1 (ABS_Y)
      Value     23
      Min    -1408
      Max     1408
    Event code 2 (ABS_Z)
      Value      0
      Min    -1408
      Max     1408
    Event code 3 (ABS_RX)
      Value     85
      Min    -1408
      Max     1408
    Event code 4 (ABS_RY)
      Value      4
      Min    -1408
      Max     1408
    Event code 5 (ABS_RZ)
      Value      0
      Min    -1408
      Max     1408
    Event code 20 (ABS_HAT2X)
      Value    344
      Min        0
      Max     1830
      Flat      30
    Event code 21 (ABS_HAT2Y)
      Value    322
      Min        0
      Max     1830
      Flat      30
Properties:
Testing ... (interrupt to exit)
Event: time 1736388382.902434, type 1 (EV_KEY), code 547 (BTN_DPAD_RIGHT), value 1
Event: time 1736388382.902434, -------------- SYN_REPORT ------------
Event: time 1736388383.078317, type 1 (EV_KEY), code 547 (BTN_DPAD_RIGHT), value 0
Event: time 1736388383.078317, -------------- SYN_REPORT ------------
Event: time 1736388385.941513, type 1 (EV_KEY), code 544 (BTN_DPAD_UP), value 1
Event: time 1736388385.941513, -------------- SYN_REPORT ------------
Event: time 1736388386.150571, type 1 (EV_KEY), code 544 (BTN_DPAD_UP), value 0
Event: time 1736388386.150571, -------------- SYN_REPORT ------------
[... 336 lines cut ...]
Event: time 1736388404.625020, type 1 (EV_KEY), code 315 (BTN_START), value 0
Event: time 1736388404.625020, -------------- SYN_REPORT ------------
Event: time 1736388406.256231, type 1 (EV_KEY), code 304 (BTN_SOUTH), value 1
Event: time 1736388406.256231, -------------- SYN_REPORT ------------
Event: time 1736388406.377230, type 1 (EV_KEY), code 304 (BTN_SOUTH), value 0
Event: time 1736388406.377230, -------------- SYN_REPORT ------------
Event: time 1736388406.685145, type 1 (EV_KEY), code 314 (BTN_SELECT), value 1
Event: time 1736388406.685145, -------------- SYN_REPORT ------------
Event: time 1736388406.894131, type 1 (EV_KEY), code 314 (BTN_SELECT), value 0
Event: time 1736388406.894131, -------------- SYN_REPORT ------------
Event: time 1736388407.642313, type 1 (EV_KEY), code 304 (BTN_SOUTH), value 1
Event: time 1736388407.642313, -------------- SYN_REPORT ------------
Event: time 1736388407.752029, type 1 (EV_KEY), code 304 (BTN_SOUTH), value 0
Event: time 1736388407.752029, -------------- SYN_REPORT ------------
Event: time 1736388409.757184, type 1 (EV_KEY), code 278 (BTN_BACK), value 1
Event: time 1736388409.757184, -------------- SYN_REPORT ------------
Event: time 1736388409.889168, type 1 (EV_KEY), code 278 (BTN_BACK), value 0
Event: time 1736388409.889168, -------------- SYN_REPORT ------------
Event: time 1736388411.517294, type 1 (EV_KEY), code 316 (BTN_MODE), value 1
Event: time 1736388411.517294, -------------- SYN_REPORT ------------
Event: time 1736388411.638325, type 1 (EV_KEY), code 316 (BTN_MODE), value 0
Event: time 1736388411.638325, -------------- SYN_REPORT ------------
^C
RetroidPocket5:~ #
`

JohnIrvine1433, 2025-01-09:
Ok thanks, try putting the attached file in the ThemeMaster folder to use manual parameters for the controls
[paramcontrols.txt](https://github.com/user-attachments/files/18366718/paramcontrols.txt)

Tom2112, 2025-01-10:
That seems to have done the trick! I didn't get an error running Thememaster on the RP5. It went straight to the menu and it's downloading themes right now!

JohnIrvine1433, 2025-01-10:
Thanks, I'll fix that bug too.

SuperBadger77, 2025-01-14:
Does Thememaster now work on the RP5?

JohnIrvine1433, 2025-01-14:
Yes you can have it working by using the manual parameters file for the input control I provided earlier. 
I'll implement the changes so you don't need to use the manual parameters in the next release.

SuperBadger77, 2025-01-15:
How far away do you think the next release is? I might try it out now though as keen. Although Rocknix just got a new release out and it says in the release notes:
Complete rebase of EmulationStation, reforked Batocera ES then applied Rocknix features and customizations. Thanks @batocera
-This includes several bug fixes and new features like theme downloader and expanded Bluetooth UI.
So seems there is a theme installer inbuilt now. I'll probably end up trying out both :-)

JohnIrvine1433, 2025-01-25:
5.2.0 released.

closed 2025-01-25 by JohnIrvine1433