JohnIrvine1433/ThemeMaster #42: not input in powkiddy x55

Kaladrigar, 2024-12-29:
at least since the rocknix version 20241120 dont undertand where is the problem and it seems like the only way to instal themes in this plataform.

JohnIrvine1433, 2024-12-29:
Strange, do you have an error message ?
If not could you confirm that you've got the following file:
/dev/input/by-path/platform-zed_joystick-event-joystick
And if so run evtest again it.

Kaladrigar, 2024-12-30:
I dont recieve an error mesage, the only thing is that is or frozen o without input, because i can´t get to the menu i have to restart the device in the themes resolution screen.

JohnIrvine1433, 2024-12-30:
try using manual parameters by putting the enclosed file in the ThemeMaster folder:

https://objects.githubusercontent.com/github-production-repository-file-5c1aeb/359454670/16149801?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAVCODYLSA53PQK4ZA%2F20241230%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20241230T124815Z&X-Amz-Expires=300&X-Amz-Signature=c6db9a9a1d27087c268672dcbf4dfdf1d2d80af05c5818ea4f6e943848ce8e8e&X-Amz-SignedHeaders=host&response-content-disposition=attachment%3Bfilename%3Dparamcontrols.txt&response-content-type=text%2Fplain

Kaladrigar, 2024-12-31:
[image]

JohnIrvine1433, 2024-12-31:
Create a file named paramcontrols.txt with the following content:
back_key = 704
start_key = 709
a_key = 304
b_key = 305
x_key = 307
y_key = 308
up_key = 544
down_key = 545
left_key = 546
right_key = 547
l1_key = 310
l2_key = 312
r1_key = 311
r2_key = 313
inputstr = /dev/input/by-path/platform-rocknix-joypad-event-joystick
handler = generic

JohnIrvine1433, 2024-12-31:
There is a bug in my implementation of the ROCKNIX joypad, it will work using manual file configuration, I'll fix the code in the next release.

Kaladrigar, 2024-12-31:
Hi man, thanks, it worked, great job, as a tip to make it more accessible you could add it to portmaster.

JohnIrvine1433, 2025-01-03:
@Kaladrigar  What do you mean by _adding it to PortMaster_?!

Kaladrigar, 2025-01-03:
Well portmaster is a plataform for ports of many kinds, usually ported games or original projects, plus I've seen a couple of tools that aren't necessarily games and I think your project fits, on their page there's contact information to see what kind of documentation you need, I think a port like yours could be useful especially for those who don't know how to access the linux files to add themes, imo you should give it a try. https://portmaster.games/faq.html

closed 2025-01-25 by JohnIrvine1433