JohnIrvine1433/ThemeMaster #43: odroid go ultra controls cant be loaded

Mattstis, 2025-01-01:
the Odroid Go Ultra with Rocknix 20241120 is giving the "controls cant be loaded. input driver not recognized." message. the device has to be hard reset after trying to open ThemeMaster.

JohnIrvine1433, 2025-01-03:
Could you tell me what you're having in /dev/input/by-path/ ?

Mattstis, 2025-01-04:
there are four files,

platform-ffd1c000.i2c-platform-rk805-pwrkey.2auto-event

platform-gpio-keys-event

platform-odroid-go-ultra-joypad-event-joystick

platform-odroid-go-ultra-joypad-joystick

Mattstis, 2025-01-04:
All four show a 0 in the far right of the screen.

JohnIrvine1433, 2025-01-06:
I'll need to implement a change in the joypad binary. In the meantime can you try using manual parameters bu creating a file named "paramcontrols.txt" within the "ThemeMaster" folder with the following content:
back_key = 704;
start_key = 709;
a_key = 305;
b_key = 304;
x_key = 307;
y_key = 308;
up_key = 544;
down_key = 545;
left_key = 546;
right_key = 547;
l1_key = 310;
l2_key = 312;
r1_key = 311;
r2_key = 313;
inputstr = /dev/input/by-path/
platform-odroid-go-ultra-joypad-event-joystick
handler = generic

Mattstis, 2025-01-07:
I created the txt file and put it in the thememaster folder. Now it is stuck on this loading screen.
[image]

JohnIrvine1433, 2025-01-07:
That's strange, the paramcontrols file is being processed before the last output.
Do you have the same issue every time you relaunch the app?

Mattstis, 2025-01-07:
Same issue anytime I launch the app and it still requires a hard reset to get out of the app.

JohnIrvine1433, 2025-01-08:
My bad, the file content was incorrect, try with this one
[paramcontrols_rocknix_ogu.txt](https://github.com/user-attachments/files/18342944/paramcontrols_rocknix_ogu.txt)

Mattstis, 2025-01-11:
After putting in the new paramcontrols.txt file in the thememaster folder I get stuck at the same screen I posted previously but now if I press a,b,x,y,l1,l2,r1,r2 or the d-pad there are signs of recognized input. Nothing from the joysticks or start and select buttons though

JohnIrvine1433, 2025-01-11:
Was the app working fine before later ROCKNIX update?

Mattstis, 2025-01-12:
I have only had the device for a few weeks now so the only version of rocknix has been the latest version.

JohnIrvine1433, 2025-01-25:
I have installed latest rocking on my device and everything works fine. Could you try to run a clean install by deleting the ThemeMaster.cfg file and the data folder if existing.
Then send me the ThemeMaster.log file.

JohnIrvine1433, 2025-02-14:
@Mattstis Just published 5.3.0 as prerelease, could you give it a try

Mattstis, 2025-02-15:
[ThemeMaster.log](https://github.com/user-attachments/files/18808279/ThemeMaster.log)

here's the themaster.log file. i tried a fresh start with the new version but still get stopped at the same point. am i missing something?

JohnIrvine1433, 2025-02-15:
Try by using this version of the script, I skipped the image loading which seems to be where you get stuck. [ThemeMaster.zip](https://github.com/user-attachments/files/18809007/ThemeMaster.zip)

Mattstis, 2025-02-16:
am i just deleting the thememaster file and replacing it with the file you provided after unzipping it?

JohnIrvine1433, 2025-02-16:
Yes

closed 2025-04-05 by JohnIrvine1433