JohnIrvine1433/ThemeMaster #44: Add support for higher resolution devices, 720p, 1080p, 1024x768, maybe more?!?

Tom2112, 2025-01-06:
I have to admit that I don't know how themes in Emulation Station work behind the scenes. So I have no idea how difficult what I'm asking is, but I love Thememaster on all of my Rocknix devices, so of course, I'd love to see it on my higher resolution devices as well. Perhaps the Rocknix creators would even consider including Thememaster in their distro. Anyway...

I realize Thememaster was made for the lower res devices, but the resolution in retro handhelds is climbing every day! My Powkiddy RGB10 MAX3 (which is 1280x720) is considered "old" in the handheld community now - even though it's only 1 year old. I also have a Powkiddy X55, which is 1280x720 as well. I have to admit, my old eyes can barely read the text (maybe you can choose a different video mode/font?)

Powkiddy is really pumping out the hi-res handhelds this year. My RGB20 Pro is 1024x768, and it would be great to have that resolution too.

With all these resolutions, would it be possible for the user to enter their resolution and Thememaster to only show themes that play nice with that res? That would be the ultimate!

Again: thanks for making Thememaster and for sharing it with the world. It really is a great thing to have on handhelds that would otherwise be locked into one theme. (I'm looking at you, Powkiddy and Rocknix! LOL) I almost mentioned a few Anbernic devices, but most of those run Knulli, which has a built-in theme downloader/handler - so no Thememaster needed. But, not to speak ill of Knulli (I love it), it doesn't give you any idea when you're downloading the theme what screen res or aspect ratio the theme is designed for. So you just have to try each one. Thememaster is much better in that regard.

Let me know what information you might need to facilitate these additions - if they're even possible. I'm not a linux guru, but I at least know how to ssh and sudo. ;)

JohnIrvine1433, 2025-01-08:
Well they are two difficulties... 
The first one is to find themes that have been specifically developed for higher resolution devices. EmulationStation themes have been initially only developed for smaller resolution, with time this has changed but most of themes out there have not been tested on high resolution. At the moment within ThemeMaster (on gallery mode) the following resolutions are managed but still most themes referenced have not been tested on the higher resolutions. I rely for the gallery mode on the following repository from Jetup which references ES themes, have a look to get an overview of what's loaded in ThemeMaster : [Emulationstation-OGA-Theme-Gallery](https://github.com/Jetup13/Emulationstation-OGA-Theme-Gallery)

The second issue is ThemeMaster's UI... As I rely on bash and terminal, UI is being managed by the OS. I cannot from the script dynamically change terminal font size or load a new font, nor change screen resolution, so I am a bit stuck there. I would like to rebuild ThemeMaster with a GUI instead of bash dialog but I can't spend enough time on it to make it come true in the near future.
For the time being the best option for you would be to manually modify the terminal configuration file to set up a bigger font size; which should do the job.

Tom2112, 2025-01-08:
Thank you! I understand about developing a GUI version. That would be pretty labor intensive.

You mentioned editing the terminal configuration file... I'm not familiar with it. Can you point me in the right direction? What's the filename and where might I find it?

JohnIrvine1433, 2025-01-09:
You should be able to do so by editing the font settings in the /etc/vconsole.conf
You should amend the FONT line (or add one). 
If needed search online for something like : change font size virtual console

Tom2112, 2025-01-10:
Thanks! I couldn't find a vconsole.conf, so I'll have to do some more digging.

closed 2025-01-18 by JohnIrvine1433