JohnIrvine1433/ThemeMaster #47: Version 5.2.0 broken on ArkOS

dani7959, 2025-01-25:
Launching ThemeMaster after the 5.2.0 update turns all buttons on the front of the device unresponsive.
Side buttons (and shutdown) still work.

RG353VS
ArkOS 24122024

Buko84, 2025-01-25:
yes, i also confir that, after update it's not working, only reset lub poweroff helps :/

Buko84, 2025-01-26:
Now it’s not even starting 🤔

JohnIrvine1433, 2025-01-26:
Hi there, can you tell me what you're having under /dev/input/by-path/ ?

davide1977, 2025-01-26:
In my case
platform-ff80000.i2c-event
platform-ordoidgo3-joypad-event-joystick
platform-ordoidgo3-joypad-joystick
platform-ordoidgo3-joypad-keys-event
platform-rk817-sound-event

Buko84, 2025-01-26:
In 353V I have files as on photo

[image]

JohnIrvine1433, 2025-01-26:
Can you try with the following

[ThemeMaster.zip](https://github.com/user-attachments/files/18550731/ThemeMaster.zip)

davide1977, 2025-01-26:
So I replaced ThemeMaster exec with the file in you zip putting it in /opt/system/tools/ThemeMaster :

-> it worked, it asked me to update, I updated -> it didn't worked anymore (handeld keys don't work)
-> I took you file again, same procedure, and now it doesn't even start anymore :(

JohnIrvine1433, 2025-01-26:
Ok sorry. 
Delete any paramcontrols.txt file in the folder. Put the ThemeMaster from the zip file. Do not update. 
Tell me if you get any input working.

davide1977, 2025-01-26:
Thanks for your answer. No, I did it again (delete paramcontrols & ThemeMaster, copied ThemeMaster exec from the zip to the same folder) : it doesn’t open anymore.

Avonavona, 2025-01-26:
And what is now the Problem? What u changed from Version before 5.2?
For me (r36s) with arkos, the controls also Not working.

Buko84, 2025-01-26:
> Thanks for your answer. No, I did it again (delete paramcontrols & ThemeMaster, copied ThemeMaster exec from the zip to the same folder) : it doesn’t open anymore.

You need to replace that thememaster.sh file that is on Tools folder, probably it’s 0KB now

davide1977, 2025-01-26:
Thanks @Buko84 do you know where I can find a working version of ThemeMaster.sh ?

Buko84, 2025-01-26:
> Thanks [@Buko84](https://github.com/Buko84) do you know where I can find a working version of ThemeMaster.sh ?

Just extract it from ArcOs image

davide1977, 2025-01-26:
I used the one from here : now it runs but it doesn’t respond anyway to keys (in this case I can’t even choose if I want to update or not)

Buko84, 2025-01-26:
> I used the one from here : now it runs but it doesn’t respond anyway to keys (in this case I can’t even choose if I want to update or not)

Ok so delete thememaster folder and copy the one from image, you will have old version that is working. We need to wait till this new one will be fixed

JohnIrvine1433, 2025-01-27:
Yes for the time being just replace with previous version

https://github.com/JohnIrvine1433/ThemeMaster/archive/refs/tags/5.1.1.zip

davide1977, 2025-01-27:
Oh yes, it works ! Thank you very much guys !

Buko84, 2025-01-28:
any news about update on working version?

dmi97, 2025-02-12:
The latest version is still broken on R36S

JohnIrvine1433, 2025-02-14:
@dmi97 Just published 5.3.0 as prerelease, can you give it a try.

SuperTaters, 2025-02-24:
> [@dmi97](https://github.com/dmi97) Just published 5.3.0 as prerelease, can you give it a try.

Hello, 5.3.0 does not work. Also, on my RG353M when trying to replace 5.2.0 with 5.1.1 in the tools folder, it still boots the 5.2.0 version. I'm not sure what I'm doing wrong.

JohnIrvine1433, 2025-02-24:
@SuperTaters what OS are you using?
Have you installed the 5.3.0 via the update function or did you do it manually ? If done manually please make sure to remove previous files especially the ThemeMaster.cfg
If done so, what's your issue with the 5.3.0 ?

SuperTaters, 2025-02-24:
> The 5.3.0 is a prerelease, the code button gives you the latest release that's why. You need to go into the releases section to download it. PS : .cfg file is initialised at first launch.

You're talking about the section underneath the green code button, correct? Because when I download the ThemeMaster.sh file that has 5.3.0 next to it and replace the same file in my tools folder, my device still boots into 5.2.0. Thank you for being patient I'm very new to this.

JohnIrvine1433, 2025-02-24:
It should say 5.3.0 when launching. 
Here is the link to the prerelease : https://github.com/JohnIrvine1433/ThemeMaster/archive/refs/tags/5.3.0.zip

SuperTaters, 2025-02-24:
> It should say 5.3.0 when launching. Here is the link to the prerelease : https://github.com/JohnIrvine1433/ThemeMaster/archive/refs/tags/5.3.0.zip

Thank you. I've copied the ThemeMaster folder and ThemeMaster.sh to the tools folder on the micro SD that goes in my TF1 slot and it's still booting into 5.2.0. I must be installing it wrong since doing the same 5.1.1 didn't change anything either. Maybe I need to uninstall ThemeMaster from the device? I've deleted the ThemeMaster.sh file and the ThemeMaster folder before copying everything over. Not sure what to do next.

JohnIrvine1433, 2025-02-25:
Doesn't the RG353M have two SD slots? If you're using the second SD slot you might install the app in the wrong folder (/rom/tools/ instead of /rom2/tools/).

SuperTaters, 2025-02-25:
> Ok, have you try installing the app in the other folder ?

I realized I had to go into the file manager on my device and launch ThemeMaster from there. Before I was launching it the normal way. 5.3.0 works now. Thank you again for being patient with me.

closed 2025-03-01 by JohnIrvine1433