JohnIrvine1433/ThemeMaster #49: ArkOS-K36 Support

AeolusUX, 2025-03-09:
Greetings,

How can I request support for devices running a custom ArkOS build? It seems that the device is not being detected. Are the controls hardware-specific rather than OS-dependent? 

Users are getting this [error](https://github.com/AeolusUX/ArkOS-K36/issues/23) when trying to open Thememaster.

AeolusUX, 2025-03-10:
lrwxrwxrwx 1 root root 9 Mar 10 09:57 platform-adc-keys-event -> ../event4
lrwxrwxrwx 1 root root 9 Mar 10 09:57 platform-ff180000.i2c-platform-rk805-pwrke                                                                                                                                                             y-event -> ../event1
lrwxrwxrwx 1 root root 9 Mar 10 09:57 platform-play_joystick-event -> ../event5
lrwxrwxrwx 1 root root 9 Mar 10 09:57 platform-play_joystick-event-joystick -> .                                                                                                                                                             ./event0
lrwxrwxrwx 1 root root 6 Mar 10 09:57 platform-play_joystick-joystick -> ../js0
lrwxrwxrwx 1 root root 9 Mar 10 09:57 platform-rk817-sound-event -> ../event3

AeolusUX, 2025-03-10:
Input driver version is 1.0.1
Input device ID: bus 0x19 vendor 0x0 product 0x0 version 0x0
Input device name: "play_joystick"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 1 (KEY_ESC)
    Event code 304 (BTN_SOUTH)
    Event code 305 (BTN_EAST)
    Event code 307 (BTN_NORTH)
    Event code 308 (BTN_WEST)
    Event code 310 (BTN_TL)
    Event code 311 (BTN_TR)
    Event code 312 (BTN_TL2)
    Event code 313 (BTN_TR2)
    Event code 314 (BTN_SELECT)
    Event code 315 (BTN_START)
    Event code 316 (BTN_MODE)
    Event code 317 (BTN_THUMBL)
    Event code 318 (BTN_THUMBR)
    Event code 464 (KEY_FN)
    Event code 544 (BTN_DPAD_UP)
    Event code 545 (BTN_DPAD_DOWN)
    Event code 546 (BTN_DPAD_LEFT)
    Event code 547 (BTN_DPAD_RIGHT)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value      9
      Min        0
      Max       18
      Flat       9
    Event code 1 (ABS_Y)
      Value      9
      Min        0
      Max       18
      Flat       9
    Event code 3 (ABS_RX)
      Value      9
      Min        0
      Max       18
      Flat       9
    Event code 4 (ABS_RY)
      Value      9
      Min        0
      Max       18
      Flat       9
Key repeat handling:
  Repeat type 20 (EV_REP)
    Repeat code 0 (REP_DELAY)
      Value    250
    Repeat code 1 (REP_PERIOD)
      Value     33

JohnIrvine1433, 2025-03-10:
I'll need to implement a change in the script to manage it. In the meantime can you try to set it manually to check that everything would be working as expected.  Create a file named "paramcontrols.txt" within the "ThemeMaster" folder with the following content:
back_key = 314
start_key = 315
a_key = 305
b_key = 304
x_key = 307
y_key = 308
up_key = 544
down_key = 545
left_key = 546
right_key = 547
l1_key = 310
l2_key = 312
r1_key = 311
r2_key = 313
inputstr = /dev/input/by-path/platform-play_joystick-event-joystick
handler = generic

AeolusUX, 2025-03-11:
I'm getting
Cannot proceed, error while executing wget:
Generic error code.

onurhazar, 2025-03-11:
@AeolusUX  fyi: I have a clone r36s and I am using your cfw - latest version.  
I deleted the existing ThemeMaster in Arkos and downloaded the latest version of ThemeMaster from: https://api.github.com/repos/JohnIrvine1433/ThemeMaster/zipball (manually copied to opt/System/Tools/)
I also created paramcontrols.txt file in ThemeMaster folder like mentioned above and it worked.

AeolusUX, 2025-03-11:
I removed it from the latest version so I didn't need to delete it on mine. I tried again now and it looks like it worked. Probably got rate limited when I tried earlier. Anyway that fixed it @JohnIrvine1433 . Thanks!

closed 2025-03-11 by AeolusUX