JohnIrvine1433/ThemeMaster #50: Provide Option to Abort install all / update all

designer2k2, 2025-04-08:
Hello,
thanks for the great ThemeMaster, i really enjoy in on my R36s!  Install was super easy and it just works, thanks!

But one thing i encountered was a 402 rate limit exceeded from GitHub when i tried to bulk install 40 Themes in one go.

That itself is ok, but i was not able to abort the installs, i had to confirm the error quite often until all the remaining Themes where done failing.

I would have liked to get an Abort Option, or any other way to stop the install all. 

Then later i did "update all" and the 402 rate limit hit again, and also here i could not abort.

So i would like to get a abort option when a error pops up during "install all" and "update all".

JohnIrvine1433, 2025-04-11:
@designer2k2 I have pushed updated files to support it but I can't get the limit on my side, could you test that it does the job for you? cf. https://github.com/JohnIrvine1433/ThemeMaster/archive/refs/heads/master.zip

designer2k2, 2025-04-12:
Hello @JohnIrvine1433,
thans for the quick work! 

The fix solves the batch install, i got the message asking me if i want to abbort the batch.

But it does not solve it for the "update all"

This is the "ThemeMaster.log" whil it happens, i clicked ok after every wget error, but would like to have also the abbort here:

```
wget error coming from BrowseMenu : Unmanaged error.\nhttps://api.github.com/repos/Jetup13/es-theme-magical-pixel/commits:
2025-04-12 12:50:02 ERROR 403: rate limit exceeded.
wget error coming from BrowseMenu : Unmanaged error.\nhttps://api.github.com/repos/Nacurutu/es-theme-ARC-S-Theme-Remix---The-RA-CFW/commits:
2025-04-12 12:50:12 ERROR 403: rate limit exceeded.
comm: file 1 is not in sorted order
comm: file 2 is not in sorted order
wget error coming from UpdateMenu : Unmanaged error.\nhttps://api.github.com/repos/Nacurutu/es-theme-ARC-S-Theme-Remix---The-RA-CFW/commits:
2025-04-12 12:50:24 ERROR 403: rate limit exceeded.
comm: file 1 is not in sorted order
comm: file 2 is not in sorted order
wget error coming from UpdateMenu : Unmanaged error.\nhttps://api.github.com/repos/Nacurutu/es-theme-ARC-S-Theme-Remix---The-RA-CFW/commits:
2025-04-12 12:50:28 ERROR 403: rate limit exceeded.
comm: file 1 is not in sorted order
comm: file 2 is not in sorted order
wget error coming from UpdateMenu : Unmanaged error.\nhttps://api.github.com/repos/Nacurutu/es-theme-ARC-S-Theme-Remix---The-RA-CFW/commits:
2025-04-12 12:50:33 ERROR 403: rate limit exceeded.
comm: file 1 is not in sorted order
comm: file 2 is not in sorted order
wget error coming from UpdateMenu : Unmanaged error.\nhttps://api.github.com/repos/Nacurutu/es-theme-ARC-S-Theme-Remix---The-RA-CFW/commits:
2025-04-12 12:50:35 ERROR 403: rate limit exceeded.
```

JohnIrvine1433, 2025-04-14:
Well that's strange, the underlying function with the abort message is the same for installation and update, you should get the same behave. 
I've been able to test it in my side by unplugging the network as you proposed and all seems fine for me...

JohnIrvine1433, 2025-04-26:
@designer2k2 Sorry for the mate reply, this was another issue... I've pushed a fix cf. https://github.com/JohnIrvine1433/ThemeMaster/archive/refs/heads/master.zip

designer2k2, 2025-04-27:
looks good @JohnIrvine1433 ! 

Thanks for the effort :-)

closed 2025-04-29 by JohnIrvine1433