JohnIrvine1433/ThemeMaster #52: Crash on startup

Nahojii, 2025-07-20:
I'm using k36 (arkos). I can start thememaster and it shows 5.4.0 but then just crashes back to desktop before the gui opens. Any idea how I can solve this?

Log:

Sun 20 Jul 2025 11:52:58 AM EDT
[H[J[3J _____ _                             
|_   _| |__   ___ _ __ ___   ___     
  | | | '_ \ / _ \ '_ ` _ \ / _ \    
  | | | | | |  __/ | | | | |  __/    
  |_| |_| |_|\___|_| |_|_|_|\___|
      |  \/  | __ _ ___| |_ ___ _ __ 
      | |\/| |/ _` / __| __/ _ \ '__|
      | |  | | (_| \__ \ ||  __/ |   
      |_|  |_|\__,_|___/\__\___|_| v5.4.0
Loading controls...
)0[1;24r[m[4l[?7h[?1000h[39;49m[39;49m[m[H[J[21drm: cannot remove '/tmp/ThemeMaster_*': No such file or directory
[0;10;1m[36m[44m[J[H ThemeMaster - System[K
 ──────────────────────────────────────────────────────────────────────────────[K
[K
[K[5;14H[1K [37m[47m┌────────────────────────────────────────────────[0;10m[30m[47m┐[0;10;1m[36m[44m[K[6;14H[1K [37m[47m│[0;10m[30m[47m[48X[6;64H│[0;10;1m[30m[40m  [36m[44m[K[7;14H[1K [37m[47m│[0;10m[30m[47m[48X[7;64H│[0;10;1m[30m[40m  [36m[44m[K[8;14H[1K [37m[47m│[0;10m[30m[47m[48X[8;64H│[0;10;1m[30m[40m  [36m[44m[K[9;14H[1K [37m[47m│[0;10m[30m[47m[48X[9;64H│[0;10;1m[30m[40m  [36m[44m[K[10;14H[1K [37m[47m│[0;10m[30m[47m[48X[10;64H│[0;10;1m[30m[40m  [36m[44m[K[11;14H[1K [37m[47m│[0;10m[30m[47m[48X[11;64H│[0;10;1m[30m[40m  [36m[44m[K[12;14H[1K [37m[47m│[0;10m[30m[47m[48X[12;64H│[0;10;1m[30m[40m  [36m[44m[K[13;14H[1K [37m[47m│[0;10m[30m[47m[48X[13;64H│[0;10;1m[30m[40m  [36m[44m[K[14;14H[1K [37m[47m│[0;10m[30m[47m[48X[14;64H│[0;10;1m[30m[40m  [36m[44m[K[15;14H[1K [37m[47m│[0;10m[30m[47m[48X[15;64H│[0;10;1m[30m[40m  [36m[44m[K[16;14H[1K [37m[47m│[0;10m[30m[47m[48X[16;64H│[0;10;1m[30m[40m  [36m[44m[K[17;14H[1K [37m[47m├────────────────────────────────────────────────[0;10m[30m[47m┤[0;10;1m[30m[40m  [36m[44m[K[18;14H[1K [37m[47m│[0;10m[30m[47m[19X[18;35H[0;10;1m[37m[44m<[33m[44m  [37m[44mO[33m[44mK  [37m[44m>[0;10m[30m[47m[21X[18;64H│[0;10;1m[30m[40m  [36m[44m[K[19;14H[1K [37m[47m└[0;10m[30m[47m────────────────────────────────────────────────┘[0;10;1m[30m[40m  [36m[44m[K[20;16H[1K [30m[40m[50X[20;67H[36m[44m[K[18;38H[39;49m[m[6;17H[30m[47mControls can't be loaded, input driver not[7;17Hrecognised. Please report for additional[8;17Hsupport.[18;38H[39;49m[m[H[0;10;1m[36m[44m[J[24;80H[39;49m[m[?1000l[24;1H
[H[J[3J

Nahojii, 2025-07-20:
[ThemeMaster.log](https://github.com/user-attachments/files/21335844/ThemeMaster.log)

JohnIrvine1433, 2025-07-20:
You're the first one reporting the use of the k36 device but as you're running arkos I wouldn't expect the app to quit, you should just get a message stating that the device is not supported.
In order to get more info, please manually activate the debug mode by editing the ThemeMaster.cfg file to "debug=on", relaunch the app and then send me the content of the newly created debug folder.

JohnIrvine1433, 2025-08-19:
It seems I totally forgot to implement a previously fixed issue...
Can you try to manually set controls as following and confirm whether it works or not. 
Create a file named "paramcontrols.txt" within the "ThemeMaster" folder with the following content:
back_key = 314
start_key = 315
a_key = 305
b_key = 304
x_key = 307
y_key = 308
up_key = 544
down_key = 545
left_key = 546
right_key = 547
l1_key = 310
l2_key = 312
r1_key = 311
r2_key = 313
inputstr = /dev/input/by-path/platform-play_joystick-event-joystick
handler = generic

JohnIrvine1433, 2025-08-21:
Fixed in v5.5.0 release.